realsenseai/librealsense #1634: Errors compiling with OpenCV examples

anshulvj, 2018-05-01:
I followed the instructions given on https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv for compiling librealsense with OpenCV examples (downloaded latest OpenCV 4.0.0 version). OpenCV examples compiles fine (when I uncheck BUILD_SHARED_LIBS and USE_EIGEN). Thereafter when I try to compile librealsense it fails with following errors:

OUTPUT:
`>f:\librealsense-5-1-18\librealsense-master\wrappers\opencv\cv-helpers.hpp(27): error C2065: 'CV_BGR2RGB': undeclared identifier
2>f:\librealsense-5-1-18\librealsense-master\wrappers\opencv\dnn\rs-dnn.cpp(56): error C3861: 'cvGetWindowHandle': identifier not found
2>f:\librealsense-5-1-18\librealsense-master\wrappers\opencv\dnn\rs-dnn.cpp(70): warning C4244: '=': conversion from 'unsigned __int64' to 'int', possible loss of data
2>f:\librealsense-5-1-18\librealsense-master\wrappers\opencv\dnn\rs-dnn.cpp(127): error C2065: 'CV_FILLED': undeclared identifier`

ERROR LIST:
`Severity	Code	Description	Project	File	Line	Suppression State
Error	C3861	'cvGetWindowHandle': identifier not found	rs-dnn	f:\librealsense-5-1-18\librealsense-master\wrappers\opencv\dnn\rs-dnn.cpp	56	
Error	C2065	'CV_BGR2RGB': undeclared identifier	rs-dnn	f:\librealsense-5-1-18\librealsense-master\wrappers\opencv\cv-helpers.hpp	27	
Error	C2065	'CV_FILLED': undeclared identifier	rs-dnn	f:\librealsense-5-1-18\librealsense-master\wrappers\opencv\dnn\rs-dnn.cpp	127	
`

Note that I have only allowed building of `rs-dnn` project in this case. If I allow the other projects (rs-grabcuts, rs-imshow, and rs-latency-tool) to compile I get same error in all of them. It's clear for some reason OpenCV hasn't compiled as it should and/or librealsense is having a hard time finding OpenCV. Note that I have added the path to the `build` folder of my successfully compiled OpenCV during Cmake Configure.

Any pointers will be helpful. Thanks in advance.

anshulvj, 2018-05-01:
SOLVED:
I had to manually address each and every error. I tried to build each and every opnecv example project separately and addressed the errors individually. The following 3 fixes worked:

1) Adding the `#include <path-to-opencv-build>/include/opencv2` folder to additional include directories in VS got rid of some of the `undeclared identifier` errors.

2) Adding `#include <path-to-opencv\modules\highgui\include\opencv2\highgui\highgui_c.h>` include file to the code which was throwing the `cvGetWindowHandle` error.

3) Changing `CV_BGR2RGB` to `COLOR_BGR2RGB`

closed 2018-05-01 by anshulvj

dorodnic, 2018-05-06:
Hi @anshulvj 
Thanks for the feedback. I just reproduced it with latest opencv version (4.0). We will update the examples ASAP