realsenseai/librealsense #1639: object tracking in 3d / pose estimation

martijnkusters, 2018-05-02:
Dear all,

I'm interested in an example of pose estimation of objects for D415 camera.
Has somebody worked already in this direction?

Best Regards,
Martijn Kusters

RealSense-Customer-Engineering, 2018-05-09:
[Realsense Customer Engineering Team Comment]
Hello,
Our SDK is focused on providing the best depth data across multiple wrappers and operating systems. We wish to grow our open source community and encourage them to do something wonderful with our products. We have no plan to add middleware libraries to our product but we may partner with 3rd parties in the future. All updates on these items will be via github and our website: http://realsense.intel.com.

GitwellAnyohub, 2018-05-09:
@martijnkusters this is sort of what I'm working on but I'm working out some kinks, particularly with acquisition parameter optimization. Once I've finished that (priority one) I can post some of my pose estimation stuff with the D415.

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
hi @GitwellAnyohub,

i am also interested if you have some example for pose estimation already, thanks.

RealSense-Customer-Engineering, 2018-07-23:
[Realsense Customer Engineering Team Comment]
going to close this ticket if no more discussion ongoing.

closed 2018-08-09 by RealSense-Customer-Engineering