realsenseai/librealsense #1642: color to infrafed pixels project have offset

Jiyun-Cui, 2018-05-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.08.15.00) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.0 }                          | 

### Issue Description
```C++
const rs2_intrinsics color_intrin = color_frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
const rs2_intrinsics ir_intrin = ir_frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics();
rs2_extrinsics color2ir_extrin = color_frame.get_profile().get_extrinsics_to(ir_frame.get_profile());
float color_face_landmark_point[4][3];
float ir_face_landmark_point[4][3];
float ir_face_landmark_pixel[4][2];
//the face landmark(four point)
float color_face_landmark_pixel[4][2]{ {faces.x, faces.y}, {faces.x, faces.y + faces.height},
    {faces.x + faces.width, faces.y},{ faces.x + faces.width, faces.y + faces.height } };
for (int j = 0; j < 4; j++) {
    rs2_deproject_pixel_to_point(color_face_landmark_point[j], &color_intrin, color_face_landmark_pixel[j], 1);
    rs2_transform_point_to_point(ir_face_landmark_point[j], &color2ir_extrin, color_face_landmark_point[j]);
    rs2_project_point_to_pixel(ir_face_landmark_pixel[j], &ir_intrin, ir_face_landmark_point[j]);
}
```
I am doing face detection. I use de/project to calculated the corresponding ir and depth pixel coordinate based on color pixel, the corresponding depth pixels coordinate are correct, but the corresponding ir pixels coordinate have offset. And I observed that the farther the distance, the smaller the offset.

dorodnic, 2018-05-03:
Hi @Jiyun-Cui 
We will have to look into it and come back with an answer

ev-mp, 2018-05-07:
Hello @Jiyun-Cui,
When deprojecting pixel to 3d coordinate note that you're using a hard-coded value of 1 (meter) which is the (presumed) Z value of the specific pixel in the source frame.
> rs2_deproject_pixel_to_point(color_face_landmark_point[j], &color_intrin, color_face_landmark_pixel[j], **1**);

When projecting a depth pixel (which has actual Z) onto a 2D frame, the projection work as designed as the Z allows to produce to calculate the exact  [x,y,z] coordinates.
In this case the projection is between RGB frame and ir frame (two 2D sources), and since the RGBframe doesn't have depth data, you're filling the missing Z value in a function call with a "rough" estimate of where the pixel "lies" in the 3D space (1 meter), which produces the error.

If the target IR frame is from the left imager (i.e. has the same origin as depth frame), then 
you may run a simple test and check the calculate error will reach minimum for pixels that actually are at  1 meter, and then start growing again.

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
hi @Jiyun-Cui,

would like to know if ev-mp's answer works for you?
thanks.

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, please feel free to open new ticket if any follow question.

closed 2018-06-11 by RealSense-Customer-Engineering