realsenseai/librealsense #1643: set_frame_callback() is not defined in librealsense2 D415

dmitryshendryk, 2018-05-03:
Camera Model                         D415 
 Firmware Version                    05.08.15.00
 Operating System & Version  Ubuntu 16
 Kernel Version (Linux Only)    4.13.0 - 36-generic                                         | 
 Platform                                   PC
 SDK Version                            2.10.4     

I'm using camera D415 and librealsense2 API. In old API there was a function for async handling the frames, but in new version i can't find this function.

**_void rs::device::set_frame_callback | ( | rs::stream | stream,|   | std::function< void(frame)> |frame_handler| )_**

I didn't find this function in new API (version 2) and to which it was replaced? Can anyone help please?

dorodnic, 2018-05-03:
Hi @dmitryshendryk 
The old function was bypassing `wait_for_frames` synchronization and producing one frame at a time.
The equivalent would be the `start` method of `sensor` class. 
Please see [sensor-control](https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L442) for example. 
Let us know if it helps.

dmitryshendryk, 2018-05-03:
Hi @dorodnic 
Yes, it helped.  Another problem, the callback doesn't called for some reason, i think doing something wrong. Can you help?

_int main(int argc, char * argv[]) try
{
    rs2::context ctx;
    rs2::device dev = ctx.query_devices().front();
    rs2::sensor sensor1 = dev.query_sensors()[0];
    rs2::stream_profile stream_profile = sensor1.get_stream_profiles().front();
    sensor1.open(stream_profile);    
    sensor1.start([&](rs2::frame f) {
        std::cout << "Test" << std::endl;
    });
}_

dorodnic, 2018-05-03:
Everything looks correct. Are you calling some blocking API after the snippet? As is the application will just finish before first frame would arrive.

dmitryshendryk, 2018-05-08:
Was missing the loop while(), it works. Thanks for help

closed 2018-05-08 by dmitryshendryk