realsenseai/librealsense #1646: I use the API provided by realsense SDK2.0, how can I use the program to drive two realsense D415?

PipgZYP, 2018-05-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2018-05-03:
Hi @PipgZYP 
Please take a look at [multicam](https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam) example.
Hopefully it will offer what you need.

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
Hello @PipgZYP, I see you added the emoji so it looks like this issue can be closed, correct?

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
Hello @PipgZYP, but this does solve your issue? Is the latter comment because you wish to run them through a hub? It is possible but it is harder to predict performance through a hub. If you have other devices I would recommend they be connected through a hub.

RealSense-Customer-Engineering, 2018-05-17:
[Realsense Customer Engineering Team Comment]
Hello @PipgZYP, it seems that this resolves your issue and I will close this however if you are still having this problem or any other one please feel free to open a new one.

closed 2018-05-17 by RealSense-Customer-Engineering