realsenseai/librealsense #1651: D415/D435 question about metadata temperature

martijnkusters, 2018-05-03:
D415/D435 question about metadata temperature

When will the metadata for temperature of the camera supported in SDK?
RS2_FRAME_METADATA_TEMPERATURE 

Best Regards,
Martijn Kusters

dorodnic, 2018-05-03:
Hi @martijnkusters 
For the D415 and the D435 temperature is not provided from firmware on frame-by-frame basis. You can still use the relevant options to query it from the sensor. 
I believe `RS2_FRAME_METADATA_TEMPERATURE` was added for a prototype D400-variant that has not been yet approved for mass-market.

RealSense-Customer-Engineering, 2018-05-09:
[Realsense Customer Engineering Team Comment]
hi @martijnkusters,

would like to confirm if you need the same as ticket #866, if so, we are still fixing on this, please stay tune.

martijnkusters, 2018-05-09:
Yes, it is the same.

Best Regards,

Martijn Kusters

dorodnic, 2018-07-11:
Let's follow-up on #866

closed 2018-07-11 by dorodnic