realsenseai/librealsense #1653: get vertices  from pointcloud in python

adarvit, 2018-05-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |  Win10  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.10.0 }                          | 

is it possible to get the vertices directly from the following code line:
`points = pc.calculate(depth)`
since the point cloud i get from the ply file (using plyfile package in python) is not the same as the xyz i get when i calculate  them by myself using rs2_deproject_pixel_to_point

zivsha, 2018-05-06:
You can:
```py
vtx = np.asanyarray(points.get_vertices())
```

adarvit, 2018-05-15:
thank you very much

closed 2018-05-15 by adarvit