realsenseai/librealsense #1656: Cannot run cv2 with python examples

anshulvj, 2018-05-03:
I'm running `python3.6` on Pycharm along with OpenCV 3.4.0 on Ubuntu 16.04 LTS
`cv2.__version__`
`'3.4.0'`

The example I'm running on this setup is the one given on https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

The code failes on the following lines:
`cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)`
`cv2.imshow('RealSense', depth_colormap)`
`cv2.waitKey(1)`

If any of the above line is uncommented the code fails with the following error:

`QObject::moveToThread: Current thread (0x190b000) is not the object's thread (0x195cd10).`
`Cannot move to target thread (0x190b000)`

`Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)`

If the three lines are commented it works fine (I am able to display the depth image using `toimage(depth_data).show()` where `depth_data` is a numpy array. So the `pyrealsense2` is working as expected. This means the problem is with OpenCV installation not working with pyrealsense. As can be seen above, I have OpenCV 3.4 running with Python 3.6 which should work according to [this post](https://stackoverflow.com/questions/46449850/how-to-fix-the-error-qobjectmovetothread-in-opencv-in-python)

I installed OpenCV as described [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv#linux)

Has anyone else faced similar issue or is aware of the solution? Any help is appreciated.

Thanks

xufeifeiWHU, 2018-05-15:
When I run the same example on the D415 depth camera, I have face the similar problem, my problem occurs on 
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
profile = pipeline.start(config)

the error like this: 
Traceback (most recent call last):
  File "/home/xufeifei/workspace/realsense/librealsense/build/wrappers/python/test2.py", line 25, in <module>
    profile = pipeline.start(config)
RuntimeError: Failed to resolve request. No device found that satisfies all requirements

have you face the the problem before?

ev-mp, 2018-05-15:
Hello @xufeifeiWHU 
The requested configuration is valid with USB3, but not with USB2 mode.
Please verify that the device is enumerated as USB3+ device using RS2_CAMERA_INFO_USB_TYPE_DESCRIPTOR info attribute.
Alternatively, open and read it in realsense-viewer.
In addition, please specify the FW version of D415.

anshulvj, 2018-05-15:
@xufeifeiWHU I had  noticed something similar but it turned out the USB cable that came with the sensor had to be pushed in all the way and the error went away.

anyone had luck with the OpenCV issue above?

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
Hi anshulvj,

Wonder if you still have issue with opencv?

I tested python3.5  (coming with Ubuntu 16.04 off Intel NUC5) with opencv (sudo apt-get install python-opencv).
After created a RealSense SDK 2.15 build w/ python wrapper enabled. I don't have issue when running opencv_viewer_example.py

Thanks,
Richard

RealSense-Customer-Engineering, 2018-11-08:
[Realsense Customer Engineering Team Comment]
Hi anshulvj,

Wonder if there is any update? Or do you still need help on this one?

Thanks!

closed 2018-11-12 by RealSense-Customer-Engineering

desert, 2020-03-16:
This has something to do with Qt, not with OpenCV.

EvanZouQ, 2020-06-17:
> When I run the same example on the D415 depth camera, I have face the similar problem, my problem occurs on
> config = rs.config()
> config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)
> config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
> 
> profile = pipeline.start(config)
> 
> the error like this:
> Traceback (most recent call last):
> File "/home/xufeifei/workspace/realsense/librealsense/build/wrappers/python/test2.py", line 25, in
> profile = pipeline.start(config)
> RuntimeError: Failed to resolve request. No device found that satisfies all requirements
> 
> have you face the the problem before?

Dear Mr Xu, I have the same error . Could you please give me some information on how you solve this problem? I have tried it several times. It always returns the same error.