realsenseai/librealsense #1657: RealSense D435 SDK run under Python Anaconda environment

mohdazly, 2018-05-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version | Windows 10 | 
| Platform                                 | PC |
| SDK Version                           | v2.10.4 | 
| Language                               | Python v2.7 & v3.6 |

### Issue Description
Is it possible to use Intel RealSense SDK Python wrappers under Python Anaconda environment?
I already followed all instruction as stated on github here https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

But i'm facing problem when i tried to run sample python project ("opencv_viewer_example.py) under my virtual anaconda environment as stated below:-

Traceback (most recent call last):
  File "opencv_viewer_example.py", line 8, in <module>
    import pyrealsense2 as rs
ImportError: Module use of python36.dll conflicts with this version of Python.  

Do you have any idea about this problem?

lramati, 2018-05-04:
Can you try compiling pyrealsense2 with anaconda as the target?
In the cmake, point pybind11 at anaconda's python executable instead of the one provided on python.org

TheMikeyR, 2018-05-04:
This is how I've compiled
```cmake ../ -DBUILD_EXAMPLES=bool:true -DBUILD_PYTHON_BINDINGS=bool:true -DBUILD_CV_EXAMPLES=bool:true -DPYTHON_EXECUTABLE=/home/mikeyr/anaconda3/envs/ENVIRONMENT_NAME_HERE/bin/python3.6```
If that doesn't work you can compile like normal and make a symlink to the library. 
Example:
```ln -s /home/mikeyr/librealsense/build/wrappers/python/pyrealsense2.cpython-36m-x86_64-linux-gnu.so /home/mikeyr/anaconda3/envs/ENVIRONMENT_NAME_HERE/lib/python3.6/site-packages/pyrealsense2.so```

mohdazly, 2018-05-08:
@lramati can you elaborate more details about pointing pybind11 at anaconda python executabel in cmake?

mohdazly, 2018-05-08:
@TheMikeyR how do i compile in Windows platform, as example you shown above i believe that using Linux platform right?

RealSense-Customer-Engineering, 2018-05-14:
[Realsense Customer Engineering Team Comment]
@mohdazly
You can use CMake GUI to check the "BUILD_PYTHON_BINDINGS" and change the "PYTHON_EXECUTABLE" to your anaconda python.exe in your environment path. For example "C:/Users/abcd/AppData/Local/Continuum/anaconda3/envs/<your ENV name>/python.exe"

After compiling python folder finish(check the attached image), you should see output files under Debug/Release folder as attachment, few things to do as below :
1. change the name of "**pybackend2.cp36-win_amd64.pyd**" to "**pybackend2.pyd**" and "**pyrealsense2.cp36-win_amd64.pyd**" to "**pyrealsense2.pyd**"
2. copy "**pybackend2.pyd**", "**pyrealsense2.pyd**" and "**realsense2.dll**" to "**DLLs**" folder. For example "C:/Users/abcd/AppData/Local/Continuum/anaconda3/envs/<your ENV name>/DLLs"
3. copy "**realsense2.lib**" to "**libs**" folder. For example "C:/Users/abcd/AppData/Local/Continuum/anaconda3/envs/<your ENV name>/libs"

Done.

carypai, 2018-05-14:
[image]
[image]

carypai, 2018-05-14:
Attach issue from official account, attached the image of activating the virtual environment and run example here
[image]

mohdazly, 2018-05-14:
@RealSense-Customer-Engineering @carypai 
As attached picture shown that there is no PYTHON_EXECUTABLE variable after I checked the configure button "BUILD_PYTHON_BINDINGS". Could you guide me on this matter. 
 
[image]

carypai, 2018-05-14:
After you check the BUILD_PYTHON_BINDINGS, please click "configure" button again, then you will see the variable.
[image]

mohdazly, 2018-05-14:
@carypai & @RealSense-Customer-Engineering thank you so much for your guidance. 

I managed to run RealSense SDK 2.0 under my anaconda virtual environment for both python 2 & 3.
Below, I'm summarizing the steps required based on my understanding, hopefully it will make it more clear for other people references. 

1. Run CMake Gui "cmake-gui.exe".
2. On "Where is the source code:", click "Browse Source" button and configure the path on librealsense, for example **"C:/Users/PCName/Desktop/librealsense"**.
3. Create "build" folder on librealsense path folder. On "Where to build the binaries:", click button "Browse Build" button and configure the path as for example **"C:/Users/PCName/Desktop/librealsense/build"**.
4. Make sure that you check on "Grouped" and "Advanced" checkbox for easy to identify each node and variables.
4. Then click "Configure" button.
5. On "Specify the generator for this project", select you visual studio version, as for example "Visual Studio 14 2015 Win64"
6. Then click "Finish" button.
7. After configuring done, on "BUILD" node, check "BUILD_PYTHON_BINDINGS" checkbox button. 
8. Then click "Configure" button again.
9. On "Ungrouped Entries", identify variable "PYTHON_EXECUTABLE", change the path to anaconda python.exe in your virtual environment path as for example 
"C:/Users/PCName/Anaconda3/envs//python.exe" or "C:/Users/abcd/AppData/Local/Continuum/anaconda3/envs//python.exe"
10. Then click "Generate" button, after all configuring and generating successfully done, click "Open Project". 
11. After Visual Studio IDE open, then start compiling/debug the solution. 

As refer to @RealSense-Customer-Engineering  
After compiling python folder finish(check the attached image), you should see output files under "librealsense/build/Debug folder as attachment inside the build, few things to do as below :

1. change the name of "pybackend2.cp36-win_amd64.pyd" to "pybackend2.pyd" and "pyrealsense2.cp36-win_amd64.pyd" to "pyrealsense2.pyd"

There are 2 methods on copying the 
1. copy "pybackend2.pyd", "pyrealsense2.pyd" and "realsense2.dll" to "DLLs" folder. For example "C:/Users/abcd/AppData/Local/Continuum/anaconda3/envs//DLLs"
2. copy "realsense2.lib" to "libs" folder. For example "C:/Users/abcd/AppData/Local/Continuum/anaconda3/envs//libs"

OR

1. You can also copy the files ("pyrealsense.2pyd" and "realsense2.dll" files) to the folder "site-packages" as for example : "C:\Users\PCName\Anaconda3\envs\EnvironmentName\Lib\site-packages".

To test that your SDK files are working well:-
1) On Anaconda prompt, activate your environment by type, "activate EnvironmentName", below scenario will be shown on the prompt. "(EnvironmentName) C:\Users\PCName>" 
2) Then navigate the python example project as below example, and then hit "Enter"
"(EnvironmentName) C:\Users\PCName>cd C:\Users\PCName\Desktop\librealsense\wrappers\python\examples"
3) Then, type python and choose one of the projects as below,
"(EnvironmentName) C:\Users\PCName\Desktop\librealsense\wrappers\python\examples>python python-tutorial-1-depth.py" 

Below picture shown the list of python sample projects for RealSense
[image]

Hereby I close the issues as I already find the solutions. 

Thank you.

closed 2018-05-14 by mohdazly

scizors, 2019-04-18:
Hello,
 I have an error while genarating C-make file
I use python 3.5.6 ,conda envirionment and windows 10 pc  
ERROR msg

Selecting Windows SDK version 10.0.10586.0 to target Windows 10.0.17134.
Info: REALSENSE_VERSION_STRING=2.19.0
Setting Windows configurations
Build for Windows > Win7
Use external libusb
CMake Error at C:/Program Files/CMake/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PythonInterp (missing: PYTHON_EXECUTABLE)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.14/Modules/FindPythonInterp.cmake:160 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  wrappers/python/third_party/pybind11/tools/FindPythonLibsNew.cmake:60 (find_package)
  wrappers/python/third_party/pybind11/tools/pybind11Tools.cmake:16 (find_package)
  wrappers/python/third_party/pybind11/CMakeLists.txt:33 (include)

Configuring incomplete, errors occurred!
See also "C:/Users/rsr5le/Desktop/realsense/librealsense-2.19.0/bud/CMakeFiles/CMakeOutput.log".
See also "C:/Users/rsr5le/Desktop/realsense/librealsense-2.19.0/bud/CMakeFiles/CMakeError.log".

JaeYoungLee79, 2022-01-03:
Everybody! Are you understanding above all? 
Is it real difficult to use realsense camera in winodw as python?
I give up!