realsenseai/librealsense #1659: rs2_create_playback_device is missing

Gnimuc, 2018-05-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.09.09.02 | 
| Operating System & Version |    MacOS  | 
| Kernel Version (Linux Only)    |  NA                                        | 
| Platform                                 | PC  |
| SDK Version                            |   master(f0caecdcb2047b968202b18bc9703bce0a74517a)                        | 

### Issue Description
A linking error occurs when using `rs2_create_playback_device`:
```C
rs2_error* e = 0;
rs2_create_playback_device("a.bag", &e);
```
```
Undefined symbols for architecture x86_64:
  "_rs2_create_playback_device", referenced from:
      _main in rs-depth.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
I find the [doc](https://github.com/IntelRealSense/librealsense/tree/master/src/media#rs2playback) uses `ctx.load_device("my_file_name.bag")` which calls `rs2_context_add_device`, so does this mean `rs2_create_playback_device` has already been deprecated? But `rs2_create_playback_device` is still in the C API [here](https://github.com/IntelRealSense/librealsense/blob/f0caecdcb2047b968202b18bc9703bce0a74517a/include/librealsense2/h/rs_record_playback.h#L69). Is there an MWE on how to use `rs2_context_add_device`?

Another question is in the [record-playback sample](https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback), I noticed new pipelines are created, configured and deleted in the process of recording, is this the best practice? Instead of stopping, recreating a new pipeline:
https://github.com/IntelRealSense/librealsense/blob/f0caecdcb2047b968202b18bc9703bce0a74517a/examples/record-playback/rs-record-playback.cpp#L89-L98

maybe it's sufficient to just create another record device via `rs2_create_record_device`? [My current implementation](https://github.com/Gnimuc/RealSense.jl/blob/882fa3901e017353b02bffddd8722126d916d147/examples/record-playback/rs-record.jl#L121-L150) looks like this:
```
# start recording
record_dev = rs2_create_record_device(dev, "a.bag", &e)
# pause recording
rs2_record_device_pause(record_dev, &e)
# resume recording
rs2_record_device_resume(record_dev, &e)
# stop recording
rs2_delete_device(record_dev)
```
and it also works. 

I'm very confused by these record-playback related APIs, I'm wondering whether should we use `pipeline` APIs or `device` APIs in this specific case. The [API Architecture doc](https://github.com/IntelRealSense/librealsense/blob/master/doc/api_arch.md) mentions two level of APIs, but it's not crystal clear to me that which API is high-level, which is not.

zivsha, 2018-05-06:
I think that `rs2_create_playback_device` is a left over from development. I don't think it was ever part of a working or supported API, so it should be removed and indeed you should use `load_device` instead.
The C++ [`context`](https://github.com/IntelRealSense/librealsense/blob/f0caecdcb2047b968202b18bc9703bce0a74517a/include/librealsense2/hpp/rs_context.hpp#L151-L167) class uses this API so you can look there for reference.

Gnimuc, 2018-05-08:
OK, thanks for your confirmation. I'll take a look at `rs2_context_add_device` and corresponding APIs this weekend.

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
Hello @Gnimuc were you able to try rs2_context_add_device? Did it resolve your issue?

Gnimuc, 2018-05-15:
It looks like using `rs2_context_add_device` along with `pipeline` will cause a crash:
```
rs2_create_context(RS2_API_VERSION, err)
rs2_context_add_device(ctx, "a.bag", err)
rs2_create_pipeline(ctx, err)  # crash
```
I guess I need to directly configure the device/sensor for streaming, I'm still learning those sensor control APIs.

Gnimuc, 2018-05-16:
I can successfully work out an example using sensor control APIs, feel free to close this issue.

closed 2018-05-17 by RealSense-Customer-Engineering