realsenseai/librealsense #1662: I use VS2017 to open realsense D415 in the way of pipeline. Why is CPU utilization so high that it can even reach 100 percent?

PipgZYP, 2018-05-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / |                                 | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.10 } | 

### Issue Description
<I use VS2017 to open realsense D415 in the way of pipeline. Why is CPU utilization so high that it can even reach 100 percent?>

dorodnic, 2018-05-07:
Hi @PipgZYP 
What language do you use?

PipgZYP, 2018-05-07:
c++

dorodnic, 2018-05-07:
Hi @PipgZYP 
If you are upgrading from older version, please check that `BUILD_WITH_OPENMP` CMake flag is disabled. 
When it is off, the overall CPU utilization should not be reaching 100%. 
There are several CPU intensive operations that we are gradually optimizing:
1. YUY to RGB conversion when using the color sensor
2. Pointcloud / Align methods that require aligning images from one viewport to another
3. Depth colorization (purely aesthetic)  
Also some examples (like `rs-measure` and some of the CV examples) add some work on top of that. 
That said, features that you don't need will not contribute to the CPU utilization - if you just enable HD depth streaming via pipeline the CPU utilization is expected to be reasonably low (8-9% on a modern PC). As you start adding more streams and processing steps, each will increase the total utilization

RealSense-Customer-Engineering, 2018-05-22:
[Realsense Customer Engineering Team Comment]
Hi @PipgZYP

Did @dorodnic answer you question with his comments? We would like to close this in five days if you have no more questions.

closed 2018-07-11 by dorodnic