realsenseai/librealsense #1663: How to control the exposure and gain and the switch of emitter by programming instead of realsenseviewer?

Destinyzc, 2018-05-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (05.08.15.00) | 
| Operating System & Version |  ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.13.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<10>.<4> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
How to control the exposure and gain and the switch of emitter by programming based on sdk 2.0 instead of realsenseviewer?

dorodnic, 2018-05-07:
Hi @Destinyzc 
Check out [controlling the laser](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser), exposure and gain can be controlled in a similar way.

Destinyzc, 2018-05-07:
@dorodnic I tried the control of laser, and it's helpful. Than you~
But for exposure and gain, can you give more information? I can't find something similar.

dorodnic, 2018-05-07:
Try using the same API calls, but replace `RS2_OPTION_EMITTER_ENABLED` with `RS2_OPTION_EXPOSURE` and `RS2_OPTION_GAIN`. 
You can use `get_option_range` function to discover the valid range of values for each option.

Destinyzc, 2018-05-07:
@dorodnic It works perfectly! Thank you very much!

closed 2018-05-07 by Destinyzc

HamdiBouchech, 2019-02-07:
yes i already turned the automatic exposure Off. 
i used the mentioned api: 
depth_sensor.set_option(RS2_OPTION_EXPOSURE, range.max)
to correctly set the exposure for the IR camera......but i would like to do the same thing for the RGB camera, is there something equivalent to: 
auto depth_sensor = selected_device.first<rs2::depth_sensor>();

to get the rgb_sensor instead of the depth_sensor ?

Thanks