realsenseai/librealsense #1664: Use SR300 on Debian stretch.

blackball, 2018-05-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   |  | 
| Operating System & Version |  Debian 9  | 
| Kernel Version (Linux Only)    |  Kernel 4.9.0                              | 
| Platform                                 | Intel |
| SDK Version                            |                         | 

### Issue Description

I would like to try SR300 on Debian 9, which has a kernel version 4.9.0. I don't prefer to upgrade the kernel directly to higher version since it will affect the installed software on my current system. I checked the patch script under ./scripts/, couldn't find a matched version for my need. Can you tell me how I could make a patch for kernel 4.9.0 safely？ 

Thanks you!

ev-mp, 2018-05-07:
Hello @blackball ,
The officially supported patches are those of Ubuntu LTS releases 14 and 16 (and in foreseeable future Ubuntu 18). There are several community-contributed patches in `scripts` directory, but none of them I believe is suited for your needs.

Therefore I would recommend you to use `patch-realsense-ubuntu-xenial.sh` as a baseline as it already covers some Debian-specific steps , modify it and then port either 4.4 or 4,10 kernel patches to Debian 9.

The things to change/track in the modified script:
1. Linux sources should point to Debian repo rather than Ubuntu ([Link](https://github.com/IntelRealSense/librealsense/blob/development/scripts/patch-realsense-ubuntu-xenial.sh#L35));
2. Make sure to get kernel sources for the exact kernel version you're running (`uname -r`), otherwise the resulted drivers may not work.
3. Modify&Merge Librealsense Ubuntu patches into the downloaded sources.([Link](https://github.com/IntelRealSense/librealsense/blob/development/scripts/patch-realsense-ubuntu-xenial.sh#L68-L76)).
4. Make sure to install the prerequisites according to the [Linux Installation guide/Backend preparation](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md#prerequisites).
5. Rerun the modified patch ones the above steps have been complete.

blackball, 2018-05-07:
Thanks! Will try it.

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
Has this issue been resolved?

RealSense-Customer-Engineering, 2018-05-31:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, please feel free to open new ticket if any question.

closed 2018-05-31 by RealSense-Customer-Engineering