realsenseai/librealsense #1667: problems after git pull and recompiling

summer1216, 2018-05-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Operating System & Version |  Linux (Ubuntu 14) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.0 - 109)                                         | 
| Platform                                 | PC  |

### Issue Description
I used librealsense for some months, and two months ago I updated once, there were no problems. And I saw many updates recently so I wanted to update again. After git pull and recompiling (followed the instruction cmake, make uninstall, make clean, make, make install...), I have a lot of troubles now. So far I can see, many examples do not work properly. Point cloud is strange displayed, live depth image is changing much slower than before, measure example crashes directly... 

I want to ask if this issue is familier, what should I do? Thank you!

dorodnic, 2018-05-07:
Hi @summer1216 
I can confirm the problem with measure and SR300. Will be fixed soon. Currently, our unit-tests mostly focus on the D400 series, but we have no intention of leaving SR300 behind. 
At a glance I didn't see anything wrong with point-cloud or synchronization but we will try to reproduce and address.

ev-mp, 2018-05-17:
Hello @summer1216 .
In the latest version [2.11.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.11.1) we fixed the rs-measure demo that now should work correctly with SR300.
Please check and confirm

summer1216, 2018-05-24:
@ev-mp  Hey, I tested it, the rs-measure demo works now with SR300. Thanks for the great work!

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
Hello @summer1216, it sounds like your issue is resolved; can this issue be closed or do you have further questions?

summer1216, 2018-05-25:
@RealSense-Customer-Engineering Hey before closing two more things please (I am sorry maybe not related to the original topic):
1.  in the _'Linux install from source manual'_, after installing cmake3 I tried to run the install_glfw3.sh. However, in the script it wants to install cmake, which has to deinstall cmake3 on my computer, it causes collision. I don't know if it is a general question or a special case. I just deleted insall cmake in the script and it works.

2. When is Realsense D435 available? I tried to buy one for a long time, but always showed something like 'unavailable'.

RealSense-Customer-Engineering, 2018-06-01:
[Realsense Customer Engineering Team Comment]
hi @summer1216,

for your question #1, I didn't see this issue before, so I think it's a special case, if you still seeing this problem, please open another ticket so we can focus on it.
for #2, do you place order on https://click.intel.com/intelr-realsensetm-depth-camera-d435.html? I see it's open for ordering now. and if you placed order already, it should ship out soon, but you better to check the click.intel.com with your order.

RealSense-Customer-Engineering, 2018-06-18:
[Realsense Customer Engineering Team Comment]
hi @@summer1216,

do you need more support on this topic?

summer1216, 2018-06-19:
@RealSense-Customer-Engineering I have contacted with custormer service about my order already. This issue can be closed. Thank you so much for every answer!

closed 2018-06-19 by RealSense-Customer-Engineering