realsenseai/librealsense #1670: Repository Issues on Linux

baptiste-mnh, 2018-05-07:
Hi,
Recently I have an issue with the repository when I do my `sudo apt-get update`: 
```
Err:8 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease
  403  Forbidden [IP: 52.218.21.49 80]
Reading package lists... Done                     
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease  403  Forbidden [IP: 52.218.21.49 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' is not signed.
```
So I can't download the components needed for the development.
I tried with two different linux distro based on Ubuntu 16.04 and 18.04.

Thanks

joebedard, 2018-05-09:
I have the same problem when I tried to install today.

baptiste-mnh, 2018-05-09:
If it's urgent, you can temporarily install it from the source files this way (on Linux) :
```
sudo apt-get install git cmake3
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
sudo apt-get install libssl-dev
sudo apt-get install libusb-1.0-0-dev pkg-config libgtk-3-dev

git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
mkdir Release
cd Release
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true.. 
make -j2 #(or -j4 if you have a quad core)
sudo make install
```
You will need some library if they missing a `sudo apt install libname-dev` is often sufficient.
For more infos you can search [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).

baptiste-mnh, 2018-05-14:
Hi,
It's starting to be urgent. I need to install intel-realsense-dfu components on my Linux and I still have this error when I update my repositories : 
```
E: The repository 'http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

```
Thanks in advance for resolving this issues as soon as possible.

feliwir, 2018-05-14:
Hitting the same issue. Please fix this

ev-mp, 2018-05-15:
Hello @baptiste-mnh , 
Did you manage to successfully [register the server's public key ](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#installing-the-packages)?
Additionally, is it possible that you're running Ubuntu distribution other than 16 ?

joebedard, 2018-05-15:
This is an issue for Ubuntu 18.04. (I don't know about 16.04.)
After successfully registering the server's public key, `sudo apt-get update` causes:
```
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease  403  Forbidden [IP: 52.218.21.217 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
```

baptiste-mnh, 2018-05-15:
I test on Ubuntu 16.04, 18.04 and Elementary OS and I got the same error than @joebedard.

For info, info this is the message I got when I register the key : 
```
$ sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
Executing: /tmp/apt-key-gpghome.DolnpYmIlw/gpg.1.sh --keyserver keys.gnupg.net --recv-key 6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: public key ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" imported
gpg: Total number processed: 1
gpg:               imported: 1
```

baptiste-mnh, 2018-05-16:
Didn't work for me @Rnandani
@ev-mp Any news? It's now down since a week...

ev-mp, 2018-05-16:
Hello @baptiste-mnh ,
The server is up and running. The packages - are [available for Ubuntu 16](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution) (xenial) only. latest version is 2.11.0.
The repo for Ubuntu 18 is in work so if you're using Ubuntu 18 you need to build for source. In that case please also await this week release to [get the proper patches for kernel 4.15](https://github.com/IntelRealSense/librealsense/pull/1702).

Two possible issues that you might hit are firewall (#1276) and proxy (https://askubuntu.com/questions/53146/how-do-i-get-add-apt-repository-to-work-through-a-proxy )
Can you check whether the download works when you're not on a corporate network ?

baptiste-mnh, 2018-05-16:
Ok, I already build from sources. I will wait the support of Ubuntu 18.04.
I just need to install intel-realsense-dfu which isn't available when I build from source.

Is there a way to get it or may be I just missed a cmake flag?

Zackjames212, 2018-05-18:
I'm having the same problem on 16.04

joebedard, 2018-05-21:
@ev-mp , firewall and proxy are not problems for me because I am using DSL connection. I am so far not successful building from source. Looking forward to 18.04 support.

ev-mp, 2018-05-23:
@joebedard , what are the issues you've encountered when building from sources ? Can you share the logs?

joebedard, 2018-05-23:
@ev-mp , No major issues, but building from source takes some time because I'm new to Ubuntu. So, I would rather wait for official apt-get support for 18.04. Thanks!

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
@baptiste-mnh The latest version v2.11.1 can support Ubuntu 14/16.04/18.04 LTS now. Please have a try to see if it resolves your issue. Thanks!

baptiste-mnh, 2018-06-05:
Hi, still not working ...
```
baptiste-mnh@baptiste-mnh-EliteBook:~$ echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list
deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main
baptiste-mnh@baptiste-mnh-EliteBook:~$ sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
Executing: /tmp/apt-key-gpghome.aTv9ZBkHky/gpg.1.sh --keyserver keys.gnupg.net --recv-key 6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1

baptiste-mnh@baptiste-mnh-EliteBook:~$ sudo apt-get update
Hit:1 http://fr.archive.ubuntu.com/ubuntu bionic InRelease
Get:2 http://fr.archive.ubuntu.com/ubuntu bionic-updates InRelease [83,2 kB]                                                                                                                                      
Ign:3 http://dl.google.com/linux/chrome/deb stable InRelease                                                                                                                                                      
Get:4 http://ppa.launchpad.net/communitheme/ppa/ubuntu bionic InRelease [15,9 kB]                                                                                                                                 
Hit:5 http://dl.google.com/linux/chrome/deb stable Release                                                                                                                                                        
Err:7 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease                                                                                                                                
  403  Forbidden [IP: 52.218.52.241 80]
Get:8 http://security.ubuntu.com/ubuntu bionic-security InRelease [83,2 kB]                                                                                    
Get:9 http://fr.archive.ubuntu.com/ubuntu bionic-backports InRelease [74,6 kB]                                                                 
Ign:10 https://dl.bintray.com/resin-io/debian stable InRelease                                                                                
Get:11 http://fr.archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [113 kB]                                                
Get:12 http://ppa.launchpad.net/communitheme/ppa/ubuntu bionic/main i386 Packages [1 180 B]                                                 
Hit:13 https://dl.bintray.com/resin-io/debian stable Release                     
Get:14 http://fr.archive.ubuntu.com/ubuntu bionic-updates/main i386 Packages [103 kB]                                       
Get:15 http://fr.archive.ubuntu.com/ubuntu bionic-updates/main Translation-en [43,9 kB]                         
Get:16 http://fr.archive.ubuntu.com/ubuntu bionic-updates/main amd64 DEP-11 Metadata [70,8 kB]                                                 
Get:17 http://fr.archive.ubuntu.com/ubuntu bionic-updates/main DEP-11 48x48 Icons [15,8 kB]                                                           
Get:18 http://fr.archive.ubuntu.com/ubuntu bionic-updates/main DEP-11 64x64 Icons [28,2 kB]                                                       
Get:19 http://fr.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [68,4 kB]                                                 
Get:20 http://fr.archive.ubuntu.com/ubuntu bionic-updates/universe i386 Packages [68,3 kB]                                              
Get:21 http://fr.archive.ubuntu.com/ubuntu bionic-updates/universe Translation-en [30,6 kB]                                          
Get:22 http://ppa.launchpad.net/communitheme/ppa/ubuntu bionic/main amd64 Packages [1 180 B]                                                
Get:23 http://fr.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 DEP-11 Metadata [117 kB]                                                            
Get:25 http://fr.archive.ubuntu.com/ubuntu bionic-updates/universe DEP-11 48x48 Icons [118 kB]                              
Get:26 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [74,1 kB]       
Get:27 http://fr.archive.ubuntu.com/ubuntu bionic-updates/universe DEP-11 64x64 Icons [186 kB]
Get:28 http://fr.archive.ubuntu.com/ubuntu bionic-backports/universe amd64 DEP-11 Metadata [5 100 B]      
Get:29 http://security.ubuntu.com/ubuntu bionic-security/main i386 Packages [64,2 kB]                         
Get:30 http://security.ubuntu.com/ubuntu bionic-security/main Translation-en [27,1 kB]
Get:31 http://security.ubuntu.com/ubuntu bionic-security/main amd64 DEP-11 Metadata [204 B]
Get:32 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [19,9 kB]
Get:33 http://security.ubuntu.com/ubuntu bionic-security/universe i386 Packages [19,9 kB]
Get:34 http://security.ubuntu.com/ubuntu bionic-security/universe Translation-en [10,2 kB]
Get:35 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 DEP-11 Metadata [2 456 B]
Reading package lists... Done                                             
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease  403  Forbidden [IP: 52.218.52.241 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

```

I also tried with : 
` echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main' | sudo tee /etc/apt/sources.list.d/realsense-public.list` (replacing xenial by bionic)
Without success... I think the issue come from the key...

ev-mp, 2018-06-05:
Hello @baptiste-mnh , Debian distribution for Ubuntu 18 (Bionic) is work in progress, and at this time Ubuntu 18 installation should be performed from source code, as [stated in the installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution).
Duplicated on #1758

baptiste-mnh, 2018-06-06:
Ok, I close this thread supposing that it works on Ubuntu 18.04.

closed 2018-06-06 by baptiste-mnh

Shreeyak, 2019-02-24:
I'm facing this issue on Ubuntu 16.04. `sudo apt update` fails:

```bash
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' doesn't support architecture 'arm64'
W: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/aptrepo xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/aptrepo/dists/xenial/main/binary-amd64/Packages  403  Forbidden
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

jinfagang, 2019-02-26:
@Shreeyak Same issue here , unable to install from ubuntu 18.04

ev-mp, 2019-02-26:
@Shreeyak ,
The debian packages are provided for X86/AMD64 architecture.  For `arm64` you need to build the lib for sources.
>N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' **doesn't support architecture 'arm64'**

We'll make sure to clarify it in the installation guide to avoid confusion. Sorry for the inconvenience.

P.S: Please make sure to reopen/open a new issue when requiring assistance - it will facilitate issue tracking and improve response times.

Shreeyak, 2019-02-26:
@ev-mp Thanks for the response. I was trying from a desktop PC, amd64 architecture, Ubuntu 16.04. I'm not sure why it tried to pick up arm64. I was following instructions in the [instructions manual](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf) for `realsense-dfu`.

I posted here since it seemed I was afflicted with the same problem as in this open issue.

I was able to resolve my issue with `apt update` failing by removing the realsense-hw packages from my sources.list. In fact, there are 2 sources in there, each of which gives a diff error (one gives the arm-64 error, one that it does not have a Release File - please see above [post](https://github.com/IntelRealSense/librealsense/issues/1670#issuecomment-466823452)).

However, surprisingly, after that I was able to install realsense-dfu with:
```
sudo apt install intel-realsense-dfu
```

NOTE: The instructions in the [instructions manual](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf) seems to have a wrong command, it asks to run
```
sudo apt install intel-realsense-du*
```
Note the '*' at the end of the command, removing that allowed me to download the package.

jinfagang, 2019-02-27:
@Shreeyak  ```
sudo apt install intel-realsense-dfu
``` works!!!

However after install how to use it?

Max-Gabriel-Susman, 2019-05-29:
@jinfagang use ' realsense-viewer '

ratmcu, 2021-01-17:
> If it's urgent, you can temporarily install it from the source files this way (on Linux) :
>
> ```
> sudo apt-get install git cmake3
> sudo apt-get update && sudo apt-get upgrade && sudo apt-get [...]

keep a space after true
`cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true .. `

ashok2811, 2021-05-26:
> If it's urgent, you can temporarily install it from the source files this way (on Linux) :
>
> ```
> sudo apt-get install git cmake3
> sudo apt-get update && sudo apt-get upgrade && sudo apt-get [...]

Add ../librealsense  after true
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true ../librealsense

niccle27, 2021-06-02:
```
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/b│
ionic/InRelease  403  Forbidden [IP: 52.218.106.194 80]                               │
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic │
InRelease' is not signed.                                                             │
```

The problem happened again and it has been like this for a week now. Could you fix it ?

mzahana, 2021-06-02:
@niccle27 I am also facing the same issue. Have you found a solution ?
@dorodnic Maybe can comment on that?

Thanks in advance.

niccle27, 2021-06-03:
i didn't find/try any solution yet, you could try to recompile from source as mentioned before. I hope they will sign the repository soon enough ...

gachiemchiep, 2021-06-03:
same problem here 

```bash
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.44.241 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

```

juandarango, 2021-06-04:
I think I found a solution:

Do not use the amazonaws.com server as described here: https://roboticslab-uc3m.github.io/installation-guides/install-realsense2.html 

But follow the intructions here, which include the intel server: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

lgulich, 2021-06-04:
This worked for me, thanks!

nairaravind, 2021-06-17:
> I think I found a solution:
>
> Do not use the amazonaws.com server as described here: https://roboticslab-uc3m.github.io/installation-guides/install-realsense2.html
>
> But follow the intructions here, [...]

Hi, I'm facing the same issue while updating my repos (similar to the errors from back in 2018 in this thread). This method doesn't seem to work in that case. Can the team please have a look at this? I'm unable to update my repos because of this.

nairaravind, 2021-06-17:
@mzahana Thanks for your reply. So just to clarify, I need to remove the current installation and rebuild from source?

mzahana, 2021-06-17:
Yes.

jonaumann, 2021-06-24:
same problem here

> Hi, I'm facing the same issue while updating my repos (similar to the errors from back in 2018 in this thread). This method doesn't seem to work in that case. Can the team please have a look at [...]

same problem here on my Jetson Nano. I have to build it from source, which kind of sucks

wwccx, 2022-03-22:
I got the same problem, however I tried the code accidentally..
 `sudo apt upgrade`
and it succeed..
I have no idea why but it's worth trying.

whizbuzzer, 2022-10-08:
I know 2 workarounds for this:

1. In the terminal, type `sudo apt install ca-certificates` (Source: https://askubuntu.com/questions/1095266/apt-get-update-failed-because-certificate-verification-failed-because-handshake)
2. Go to **"Software and Updates”** -> **“Other Software”** -> click on the troublemaker(s), and click on remove
(Source: https://www.linuxfordevices.com/tutorials/ubuntu/remove-an-apt-repository-ubuntu)

wwccx, 2022-10-08:
你好，你的邮件已收到。我会尽快处理，谢谢。

ibtehajali67, 2023-10-11:
> keep a space after true `cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true .. `

its work for me, Thank you