realsenseai/librealsense #1671: Error compiling install target on macos

wingillis, 2018-05-08:
When compiling the realsense sdk for the `install` target or the `ALL_BUILD` target using xcode on macos I get an error for this line of code:
https://github.com/IntelRealSense/librealsense/blob/master/src/types.h#L821

Which says: `Too few arguments to function call, expected 4, have 3`

How should I fix this issue?

dorodnic, 2018-05-08:
Hi @wingillis 
I don't see the error with Xcode on a MacBook Pro, and the [CI](https://travis-ci.org/IntelRealSense/librealsense/jobs/376221039) is also not raising an issue.
Are there any special CMake flags you enable? Perhaps there is an older version of the SDK installed (using `sudo make install` at some past time)

wingillis, 2018-05-08:
Thanks for responding so quickly! I noticed the CI was not raising an issue. When I build just the realsense2 library, the build passes. I think it's breaking while trying to compile the `realsense-viewer`. 
Here are the CMake flags I used (I've run both commands below and the subsequent build in Xcode failed):
`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DBUILD_PYTHON_BINDINGS=bool:true -DBUILD_RS400_EXTRAS=true -DHWM_OVER_XU=false -G Xcode`

`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DBUILD_PYTHON_BINDINGS=bool:true -DHWM_OVER_XU=false -G Xcode`

I ran these in an empty `build` directory

dorodnic, 2018-05-08:
I'm almost positive you have older version installed. Please check `/usr/local/include`,  `/usr/local/lib` and `/usr/local/bin` for realsense stuff and delete it. 
Please note that `rs2_frame_processor_callback_ptr` delegate used to receive 4 arguments prior to 2.11, but receives only 3 now.

wingillis, 2018-05-09:
I removed the realsense related code from `/usr/local/include` and `/usr/local/lib` and the compilation worked! Thanks for the help.

closed 2018-05-09 by wingillis