realsenseai/librealsense #1672: Python - How to use post processing APIs provided in python wrapper

zidanehuang001, 2018-05-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415 / 435
| Firmware Version                   | 5.9.9.2 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.11.0                         | 

### Issue Description
I'm using python wrapper for D415, I noticed in this 2.11.0 version you provide post processing APIs in python wrapper. Could you provide an example on how to use them?

zivsha, 2018-05-08:
The below code was not tested, and might contain some error, but this could help you get started

```py
import pyrealsense2 as rs
pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_stream(rs.stream.Depth, 640, 480, rs.format.Z16, 30);
pipe.start(cfg)
# Declare filters
dec_filter = rs.decimation_filter ()   # Decimation - reduces depth frame density
spat_filter = rs.spatial_filter()          # Spatial    - edge-preserving spatial smoothing
temp_filter = rs.temporal_filter()    # Temporal   - reduces temporal noise
frames = pipe.wait_for_frames()
depth_frame = frames.get_depth_frame()
filtered = dec_filter.process(depth_frame)
filtered = spat_filter.process(filtered)
filtered = temp_filter.process(filtered)
# do something with 'filtered' which is a rs.frame
```

adarvit, 2018-05-15:
hey, i get the following error when i am trying to declare those filters

"AttributeError: module 'pyrealsense2' has no attribute 'decimation_filter'

i tried to upgrade pyrealsense2 but it didn't solve the problem

zidanehuang001, 2018-05-16:
@lramati thank you

closed 2018-05-16 by zidanehuang001

lihk11, 2019-07-03:
@lramati  Is it possible to set persistency index for temporal filter using Python?

lramati, 2019-07-03:
It should be possible.
The options API should suffice, although we've had reports of it not
working correctly with filters. If that's the case, you can use the
extended constructor which takes smoothing alpha/delta parameters and also
the persistence control.

lihk11, 2019-07-04:
@lramati  Thanks your reply first!

lihk11, 2019-07-04:
@lramati Thanks for your timely reply!
I could find `rs.option.filter_smooth_alpha / filter_smooth_delta` which configures the alpha/delta parameter, but it seems that there isn't a "persistence control" attribute... Could you point out the exact attribute if it exists?

lihk11, 2019-07-04:
I just figure it out.