realsenseai/librealsense #1676: Why realsense 2.0  can't be compiled on visual studio 2013 update 5?

lishirui, 2018-05-08:
I try to compile realsense 2.0, but failed. It only can be compiled in legacy branch which I think is realsense 1.0. Am I right?

RealSense-Customer-Engineering, 2018-05-10:
[Realsense Customer Engineering Team Comment]
hi @lishirui,

I think we just verified Visual Studio 2015 and 2017, if you don't really need the legacy branch, would suggest to use 2015 or 2017 version to compile.

dorodnic, 2018-05-13:
Hi @lishirui 
We decided on Visual Studio 2015 since it more fully complies with the C++11 set of features used in the SDK. With VS 2013, I recall often getting "Internal Compiler Error" on valid code. 
This should not prevent you from using RealSense with Visual Studio 2013, since you can use the precompiled version ([Intel.RealSense.SDK.exe](https://github.com/IntelRealSense/librealsense/releases/download/v2.11.0/Intel.RealSense.SDK.exe))

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
hi @lishirui,

do you need more support on this topic?

lishirui, 2018-06-19:
Yes. It would be great if Realsense 2.0 can support  VS2013. Thank you for your reply.

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
hi @lishirui,

I don't think we have plan to support VS2013,and I think dorodnic's suggestion should work for you on VS2013, please let us know if any issue with precompiled version, thanks.

closed 2018-07-11 by dorodnic