realsenseai/librealsense #1678: BUG - Intel SR300 - Gesture control?

shamun, 2018-05-09:
https://software.intel.com/en-us/articles/building-gesture-recognition-web-apps-with-intel-realsense-sdk
[image]

I have SR300 camera which was working with old Javascript SDK (since 2015 till 2017). After windows 10 update now it never work anymore because i have removed deepfreeze. is it a BUG?

How to use the gesture control in 2018? 
(OLD SDK Javascript not working anymore)

RealSense-Customer-Engineering, 2018-05-10:
[Realsense Customer Engineering Team Comment]
@shamun,
dose the win hello also not functional/not able to setup (the SR300 need to be installed properly)
please find below to see if that can help on your case
https://www.intel.com/content/www/us/en/support/articles/000026184/emerging-technologies/intel-realsense-technology.html

RealSense-Customer-Engineering, 2018-05-22:
[Realsense Customer Engineering Team Comment]
Hi @shamun,
could below link helpful on your case? that to ensure the SR300 DCM installed properly
https://www.intel.com/content/www/us/en/support/articles/000026184/emerging-technologies/intel-realsense-technology.html

shamun, 2018-05-22:
I followed the steps, but its failing to install the software DCM?
[image]

RealSense-Customer-Engineering, 2018-05-23:
[Realsense Customer Engineering Team Comment]
@shamun,
please go to Control Panel\Programs\Programs and Features
(1) uninstall the" intel realsense Depth Camera manager SR300"
(2) reboot the PC
(3) with SR300 plug into USB3 port, install the driver again (use .inf to install the driver)
      -go to the driver package there should be a inf file named :RealSenseSR300Depth.inf
      -right click the inf file, then chose install from the pop up menu

shamun, 2018-05-23:
Uninstalled, rebooted,retry but it wont install. Where do i get the exact files to make it work?

[image]

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
@shamun,
The intel_rs_dcm_sr300_3_3_27.5718 is the latest SR300 driver which also post in windows update server
that you can use windows update to install it after uninstall the SR300 driver from program/features
that OS will detect the camera type then install the corresponding driver (f200 or SR300)

F200 peripheral (made by creative with dual mic ) is old version of SR300

Uninstalled, rebooted,retry but it wont install --- no virtual camera enumerated under device manager imaging class?

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
Hi Shamun,
the SDK is not maintained for a while.
You could file another ticket with proper subject to get more inputs

closed 2018-06-25 by RealSense-Customer-Engineering