realsenseai/librealsense #1682: Adding realsense2 to my existing C++ project through CMake

adarshmuthappa, 2018-05-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.09.09.02 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.4                         | 

### Issue Description
I am stuck trying to add realsense2 to my C++ project in my CMake file. I have tried to do `find_package(realsense2 REQUIRED)` but CMake throws an error saying - 

Could not find a package configuration file provided by "realsense2" with
any of the following names:

realsense2Config.cmake
realsense2-config.cmake

Is there an easy way to add realsense SDK to my existing VS 2017 C++ project?

mutp, 2018-05-09:
Maybe I posted this question too soon, but I found a solution.
https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

The CMakeLists file here lets me add realsense to my existing C++ project through CMake.

closed 2018-05-09 by mutp

gayanbrahmanage, 2018-10-13:
Try this
https://github.com/IntelRealSense/librealsense/issues/825#issuecomment-429065088