realsenseai/librealsense #1684: Docker image for using SR300

blackball, 2018-05-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300  } | 
| Firmware Version                   | | 
| Operating System & Version |  Linux | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | |
| SDK Version                            |                        | 

### Issue Description

I wonder if there's any ready to use docker image for using SR300 and D400 series. If I need to do it myself, how do I enable the realsense device in the container ? Thanks!

RealSense-Customer-Engineering, 2018-05-14:
[Realsense Customer Engineering Team Comment]
hi @blackball,

can you please explain more about docker image you mean?
thanks.

blackball, 2018-05-14:
I wonder how I can use SR300 inside a docker container, because my host OS doesn't have a matched kernel with SR300's driver.

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
Hello @blackball,

There is no ready to use docker image for RealSense Cameras.  Docker does not provide access to the USB devices on the host system.  If you insist on creating a virtual environment you may use a virtual machine.  VMware Workstation Player works in emulating USB 3 layers, just be sure to add USB 3 devices to image prior to booting.  Also note virtual machines are NOT recommended by RealSense team as this additional layer can impact the camera performance.

 See this link for more details:  https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A#q-im-using-a-virtual-machine-and-the-camera-is-not-working

Thank you,

Nate W.

closed 2018-05-24 by RealSense-Customer-Engineering

blackball, 2018-05-25:
@RealSense-Customer-Engineering 

"Docker does not provide access to the USB devices on the host system." This is not true.

blackball, 2018-05-30:
If you search on google, you can immediately find docker image for R200: https://hub.docker.com/r/mjenz/ros-realsense/.

njwardx, 2018-06-06:
Blackball,
I want to thank you for your question, we were able to create a docker image with LibRealSense samples.  
You can try following directions posted here, along with pulling our docker container:  https://hub.docker.com/r/speechtesting11/ubuntu16.04_lrs3/
Docker pull command:  sudo docker pull speechtesting11/ubuntu16.04_lrs3

SR300 was tested and is able to run with app RealSense Viewer along with D400 series cameras.
Please let us know if you have any other questions.
Thank you.

blackball, 2018-06-07:
@njwardx 

Thanks for working on this issue. I followed steps in the docker hub page. The RGB camera worked out of the box, but not the depth part. 

> Backend in rs2_open_multiple(sensor:0x7f6cc4000f80, profiles:0x7f6cc4317470, count:1):
> The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a > patch!
> Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device

Seems I need to patch the kernel.

RealSense-Customer-Engineering, 2018-06-07:
[Realsense Customer Engineering Team Comment]
hi @blackball,

yes, from the message, you need to patch your kernel.

Transmitt0r, 2018-06-19:
@njwardx can you provide the Dockerfile for the linked image? I'm really curious how the USB device is passed to the container!

njwardx, 2018-06-21:
@Transmitt0r 
I didn't use a dockerfile I simply installed docker on Ubuntu and pulled Ubuntu base image and built LRS SDK with these commands:

sudo docker pull ubuntu

xhost local:root 

sudo docker run ubuntu

sudo docker run -it ubuntu

Followed instructions here:  https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md 

Thank you.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
thanks @Transmitt0r  for sharing.
hi @Blackball, do you need more support on this topic?

blackball, 2018-08-05:
@RealSense-Customer-Engineering No, Thanks.

lauer2356, 2018-11-08:
> Here is the Dockerfile I built, if someone stumbles over this.
> I use it to run a Python 3.6 application with a GUI
> https://gist.github.com/Transmitt0r/698dac6946fd43be19b8f799f82f326e

@Transmitt0r - do you still have the dockerfile available? Looks like the link above is dead.

kmonachopoulos, 2019-01-07:
Does not work on macos. Still cannot identify the device ...

HugoHJH, 2020-04-09:
> @blackball
> Blackball,
> I want to thank you for your question, we were able to create a docker image with LibRealSense samples. 
> You can try following directions posted here, along with pulling our docker container:  https://hub.docker.com/r/speechtesting11/ubuntu16.04_lrs3/
> Docker pull command:  sudo docker pull speechtesting11/ubuntu16.04_lrs3
> 
> SR300 was tested and is able to run with app RealSense Viewer along with D400 series cameras.
> Please let us know if you have any other questions.
> Thank you.

Would it be possible for you to share the parts of the Dockerfile that allowed to you install the RealSense2 library please?

lucbettaieb, 2020-12-09:
For future travelers....

 - Add apt sources for librealsense packages
 - On the host, install librealsense2-dkms
 - In the container, install everything else
 - Run the container (docker run) with --priveleged and -v /dev/:/dev/ with /bin/bash or whatever app you want to run at the end
 - If you're in a bash shell, try `rs-fw-update -l`
Profit(??)