realsenseai/librealsense #1685: Cannot modify stream resolution with pipeline configuration

visez, 2018-05-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    |                                | 
| Platform                                 | Mac  |
| SDK Version                            |  { 2.2.11 }                          | 

### Issue Description

I am trying to set the rgb and depth resolution by passing a rs2::config object to the pipeline object. However, when I launch the program, I get the following error:

> libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: Frame format is not supported yet!

Here is the minimal code to reproduce the problem:

```    
    //realsense configuration options
    rs2::colorizer color_map;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 360);
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 360);
    
    rs2::pipeline pipe;
    pipe.start(cfg);
    cv::Size rgbSize, depthSize;
    
    const auto depthWindowName = "Depth";
    const auto rgbWindowName = "RGB";
    cv::namedWindow(depthWindowName, cv::WINDOW_AUTOSIZE);
    cv::namedWindow(rgbWindowName, cv::WINDOW_AUTOSIZE);
    
    while (cv::waitKey(1) < 0 && cvGetWindowHandle(depthWindowName) && cvGetWindowHandle(rgbWindowName))
    {
        //get frames and align
        rs2::frameset data = pipe.wait_for_frames();
        rs2::frame depth = data.get_depth_frame();
        rs2::frame rgb = data.get_color_frame();
        if (!depth)
            continue;
        
        //get opencv matrices
        cv::Mat depthMat = frame_to_mat(color_map(depth));
        cv::Mat rgbMat = frame_to_mat(rgb);
        
        // show images
        imshow(depthWindowName, depthMat);
        imshow(rgbWindowName, rgbMat);
    }
```

dorodnic, 2018-05-10:
Hi @visez 
You can most likely work around this problem by using the full overload of `enable_stream` with rest of the parameters. That said, this seems suspicious behavior and likely to be a bug, we will reproduce and update.

visez, 2018-05-10:
@dorodnic thank you - just tried with the following code but I get the same problem unfortunately:

```
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, -1, 640, 360, rs2_format::RS2_FORMAT_ANY, 0);
    cfg.enable_stream(RS2_STREAM_DEPTH, -1, 640, 360, rs2_format::RS2_FORMAT_ANY, 0);

    rs2::pipeline pipe;
    pipe.start(cfg);
```

visez, 2018-05-10:
@dorodnic Just checked the firmware version - 05.08.15.00. Could that be the reason?

dorodnic, 2018-05-10:
Could you please check using the Viewer if the camera is detected as USB3? 
I tried the original code locally and everything worked. I suspect the problem can be USB connection, or perhaps the firmware version. Unfortunately, we still don't have Mac OS firmware update tool (only for Windows and now also Ubuntu I believe)

visez, 2018-05-11:
Updated the firmware to 5.9.9.2 from a windows PC, and the realsense is detected as usb3.0, the problem however still persists
[image]

dorodnic, 2018-05-13:
Could you please make sure that you can manually select 640x360 from the RealSense Viewer?

visez, 2018-05-15:
Silly mistake - by putting RS2_FORMAT_ANY in the call for the color stream it got automatically assigned to Y16. Switching the code to the following solved the problem

```
cfg.enable_stream(RS2_STREAM_COLOR, -1, 640, 360, rs2_format::RS2_FORMAT_RGB8, 0);
```

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
hi @lramati,
so your problem is solved?

RealSense-Customer-Engineering, 2018-05-25:
[Realsense Customer Engineering Team Comment]
that's right, thanks @lramati.

hi @visez,

would like to confirm if your problem is solved?
thanks.

visez, 2018-05-29:
Thank you, mark it as solved.

closed 2018-06-06 by RealSense-Customer-Engineering