realsenseai/librealsense #1686: Repeated frames passed by D435 (bug?)

chrelli, 2018-05-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Ubuntu 16.04 LTE   | 
| Kernel Version (Linux Only)    |  4.13.0-39-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |   2.10.4 (python wrapper) |

### Repeated frames passed by D435

When i read from a d435 through the python wrapper (using `pipeline.wait_for_frames()`), I sometimes receive the same frame twice (same frame number and hw time stamp). 

I can check and reject repeated frames. But is this a bug, or is that the expected behavior?

chrelli, 2018-05-10:
It seems to be only in the beginning of recordings and 'warming up' the camera seems to get rid of the issue. Will update later.

chrelli, 2018-05-10:
Nope, it can happen in the middle of longer recordings (15 min) as well. Warming up doesn't fix it.

dorodnic, 2018-05-13:
Hi @chrelli 
Yes, this is the expected (or at least possible) behavior. This can happen when one of the streams drops frames (in that case we will return last known frame from that stream).
If you are building the SDK from source, please make sure to build with `-DCMAKE_BUILD_TYPE=Release`. You can also try reducing the total resolution.

RealSense-Customer-Engineering, 2018-05-22:
[Realsense Customer Engineering Team Comment]
hi @chrelli,

would like to know if build with `-DCMAKE_BUILD_TYPE=Release` works for you?
thanks.

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, please feel free to open new ticket if any follow question.

closed 2018-06-11 by RealSense-Customer-Engineering

chrelli, 2020-02-20:
No, -DCMAKE_BUILD_TYPE=Release does not fix it. I had already compiled the release version.

Just so that I understand the expected behavior. Is the following correct?

When a frame is dropped, the last known frame, frame number and hardware time stamp is passed (so it's repeated). OK. But when the next new frame arrives, is the hardware time stamp counter is just updated, as if the dropped frame never happened?

I mean, if record at 60 fps, the time difference in the hardware time stamps is ~16 ms.

Let's say that I drop 2 frames, then the difference in time stamps will be like this:
[16, 16, 16, 0, 0, 16, 16, 16, 16]

Even though, in reality, it should be something like:
[16, 16, 16, 0, 0, 48, 16, 16, 16]

Since, in the real world, 3 frame cycles has elapsed while the camera was dropping frames.

I mean, does a dropped frame just mean that the camera misses one cycle, at the same frame rate?

ndaley7, 2024-10-27:
@RealSense-Customer-Engineering  is there any answer on the question posed here?

https://github.com/IntelRealSense/librealsense/issues/1686#issuecomment-588597912

MartyG-RealSense, 2024-10-28:
Hi @ndaley7  I do not have a comment to add to https://github.com/IntelRealSense/librealsense/issues/1686#issuecomment-588597912

https://github.com/IntelRealSense/librealsense/issues/5885#issuecomment-590502775 advises that if RGB frames have had their FPS fixed by disabling the RBB option Auto Exposure Priority then frame drops can occur if the computer's CPU is experiencing a high processing load.