realsenseai/librealsense #1687: Repeatedly changing exposure of d435 brings down a camera

DrCodeWizzard, 2018-05-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.9.9.2 | 
| Operating System & Version |  Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  tegra ubuntu 4.4.38                                        | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.9.1 }                          | 

### Issue Description
We are running a system with two d435 cameras connected to an NVIDIA TX2. We discovered that repeatedly changing exposure on one camera will bring the second camera down.

We used the following bash script to invoke ros param server to change the exposure settings:

`#!/bin/bash`
`realsense2_camera="/sensors/camera_front_forward/realsense2_camera"`
`while :`
`do`
`#loop infinitely`
`rosrun dynamic_reconfigure dynparam set $realsense2_camera rs435_depth_exposure 200`
`sleep 5`
`rosrun dynamic_reconfigure dynparam set $realsense2_camera rs435_depth_exposure 30`
`sleep 5`
`done`

With the latest firmware released by Intel we can get this script to bring down a camera after about 5 hours run time ( interestingly it brings down the second camera down, and not the one on which we change the exposure rates)

With a previous version of firmware we can reproduce this in about two hours with the script above.
  

We have applied the UVC Linux drivers to the NVIDIA Tegra kernel patch as per installation instructions.

DrCodeWizzard, 2018-05-16:
I used a powered USB 3.0 Amazon Basics hub to connect two cameras to TX2 board. We also have a custom board (for our product) which exposes 5 USB 3.0 ports. I was able to reproduce this behavior on both setups.

We are running one camera with RGB set to 640x480 @ 15 Hz. The second camera doesn't have color enabled at all. Depth is captured at 424 x 240 @ 30Hz. We change exposure settings for the camera which has RGB enabled.

RealSense-Customer-Engineering, 2018-05-21:
[Realsense Customer Engineering Team Comment]
@DrCodeWizzard,
basically, we recommend to use dedicated USB3 for D435 to ensure there sufficient bandwidth  especially for high fps and high resolution.
when you said the second device down, is it not enumerated any more?
is there lsusb -t or dmesg for reference?

by the way, please we have  newer FW,  v5.9.11, which has fix for device recognize issue and some control related issue
https://downloadcenter.intel.com/download/27792/Latest-Firmware-for-Intel-RealSense-D400-Product-Family

DrCodeWizzard, 2018-05-21:
Yes, the second device is no longer enumerated. I will check the new firmware.

Also yes, I am aware of the USB3 recommendation and that is why our production board exposes 5 dedicated ports.

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
@DrCodeWizzard,
could the new FW help on your issue?

DrCodeWizzard, 2018-06-06:
The latest firmware appears to make the camera more stable to crashes due to changing exposure levels. I have been able to run the script for several days without a break.

closed 2018-06-27 by RealSense-Customer-Engineering