realsenseai/librealsense #1689: ds5lr camera

martijnkusters, 2018-05-14:
Question: Does somebody know when the ds5lr camera will be available. 
It is a long range camera with a larger baseline than the d400-serie.

Best Regards,
Martijn Kusters

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
hi @martijnkusters,

let me check on this for you, and I assume you are checking for peripheral not bare module, right?

RealSense-Customer-Engineering, 2018-05-16:
[Realsense Customer Engineering Team Comment]
hi @martijnkusters,
per checked, we don't have plan for DS5lr peripheral at this point, sorry for this.

TheMikeyR, 2018-05-16:
what about the bare module? @RealSense-Customer-Engineering

RealSense-Customer-Engineering, 2018-05-21:
[Realsense Customer Engineering Team Comment]
hi @TheMikeyR,

do you have contact with intel sales?
if no, would you provide your contact so we can reach you and check for your need?

TheMikeyR, 2018-05-23:
I'm not much for giving up my email address in a public forum, is it possible to find the contact info by looking me up on designintools by username? If so my username is MikeyR.

RealSense-Customer-Engineering, 2018-05-23:
[Realsense Customer Engineering Team Comment]
hi @TheMikeyR,

sure, let me try with that approach.

TheMikeyR, 2018-05-23:
@RealSense-Customer-Engineering a representative from Intel have contacted me now, thanks!

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
hi @TheMikeyR,

great, cheers.

hi @martijnkusters,

do you have any follow up question for this topic?

martijnkusters, 2018-05-24:
No, you can close this topic

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
thanks and please feel free to create a new ticket if any question

closed 2018-05-24 by RealSense-Customer-Engineering