realsenseai/librealsense #1692: possibly missing const?

martinakos, 2018-05-14:
The template function device::first  in line 51 [here](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_device.hpp)
may need a const? 

See [this](https://stackoverflow.com/questions/50328701/const-correctness-advice) stackoverflow question.

martinakos, 2018-05-14:
Also, possibly missing const in function frameset::get_color_frame at line 678 [here](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_frame.hpp)

RealSense-Customer-Engineering, 2018-05-18:
[Realsense Customer Engineering Team Comment]
@martinakos
I created PR on behalf of you as below.
https://github.com/IntelRealSense/librealsense/pull/1726

closed 2018-06-14 by RealSense-Customer-Engineering