realsenseai/librealsense #1693: IR didn't work anymore

baptiste-mnh, 2018-05-14:
### Issues
Hi,
Since this morning, 3 of my 6 cameras have their IR captors broken.
I tried to update them to the last firmware (5.9.11) but it didn't change.
It is surprising that it happens at the same time. So are they broken or this could be an software issues?
(It's always the same cameras which can't stream there IR view)

### Error message 
```
Invalid Value in rs2_get_option_range(options:0x55fea0a9bd90, option:Stereo Baseline, min:0.0001, max:0.01, step:1e-06, def:0.001):
OpCodes do not match! Sent 21 but received -6!
```
### Screenshot
[image]
Same issue on Ubuntu 18.04 and Window 10

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
hi @baptiste-mnh,

from previous experience, seems your FW somehow damaged, can you please try the calibration tool,https://downloadcenter.intel.com/download/27415/?v=t
 with Intel.Realsense.CustomRW.exe -g to recover the calibration table and see if it recovered.
maybe same issue to #1631

baptiste-mnh, 2018-05-15:
Hi, I tried to calibrate them but I got this message : 
```
Error: calibration tables on this device are not supported. Please try again whith the lastest Dynamic Calibration Software.
If you are using latest software, please re-calibrate the devcie with the latest OEM or Technician Calibration tools and try again
```

RealSense-Customer-Engineering, 2018-05-17:
[Realsense Customer Engineering Team Comment]
hi @baptiste-mnh,

would like to confirm if the message is from command Intel.Realsense.CustomRW.exe -g?
thanks.

baptiste-mnh, 2018-05-17:
My bad, I was on linux and I used the Dynamic calibration tool. They works like a charm now, thanks!
Is there a linux equivalent command?

RealSense-Customer-Engineering, 2018-05-18:
[Realsense Customer Engineering Team Comment]
hi @baptiste-mnh,

great to hear that your issue fixed.
no, the dynamic calibration tool have only windows version so far.

baptiste-mnh, 2018-05-18:
Ok thanks !

closed 2018-05-18 by baptiste-mnh