realsenseai/librealsense #1697: Multicam D415 gives me Device or resource busy

jgomezlopez, 2018-05-14:
Good evening,

I'm trying to get depth images from several D415 at 640 x 480 and 6 fps.

When I try to get images from one camera everything works fine.
When I try to get images from two cameras sometimes works fine but most of the times give the following error:

terminate called after throwing an instance of 'rs2::backend_error'
what():  xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy

When I try to get images from 3 cameras always get the mentioned error.

I'm running in a Ubuntu 16.04 with a 4.10 Kernel in a NUC 6 i3 and NUC 7 i3 using librealsense 2.11.0
 
When I get the error I also get the following info with dmesg command

[ 6966.431841] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6966.439666] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6966.444264] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6966.482433] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6966.490424] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6966.494982] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6968.746609] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6968.757428] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6968.780008] uvcvideo: Non-zero status (-71) in video completion handler.
[ 6968.780090] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6968.797231] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6968.808122] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6971.097573] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6971.116495] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6971.121687] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6971.148123] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6971.167287] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6973.014126] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6973.125005] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6973.147416] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6973.152478] uvcvideo: Failed to query (SET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6973.175654] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6973.198098] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 6974.610032] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

Thanks in advance for helping me

RealSense-Customer-Engineering, 2018-05-30:
[Realsense Customer Engineering Team Comment]
hi @jgomezlopez,

would like to know how you connect the camera to PC? with USB hub?
we tested in NUC i3 for 1.5 hr and didn't see this issue.
and are you testing with multi-cam sample or other app?
thanks.

jgomezlopez, 2018-06-06:
I was testing with other app. Now I tested with rs-multicam example and it works.

closed 2018-06-06 by jgomezlopez

locdoan12121997, 2020-02-27:
@jgomezlopez Hi, how did you solve the problem, I get the same message as you with command dmesg