realsenseai/librealsense #1698: D415 High Accuracy Mode, black color quality too low

dmitryshendryk, 2018-05-15:
Camera Model D415
Firmware Version 05.08.15.00
Operating System & Version Ubuntu 16
Kernel Version (Linux Only) 4.13.0 - 36-generic |
Platform PC
SDK Version 2.10.4

I'm doing the reconstruction using D415 camera. I'm trying to improve the quality of the model, so now use High Accuracy Mode in librealsense2. 
In High Density Mode, black color better, but the structure not good, not works for me.

But now the quality of black color is too low. How to improve black color detection and don't lose structure quality?
[image]
[image]

dmitryshendryk, 2018-05-15:
Sorry, the last image is wrong. Updated correct one.
[image]

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@dmitryshendryk
For high accuracy, it will generate data for only high confident pixel and ignore low confident one with black color. But few suggestions might improve the quality.

1. increase the laser power.
2. disable auto exposure, from the image, I can tell the IR image is pretty dark.
3. Try to tune the image quality with the suggestion in side the document. https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf

dmitryshendryk, 2018-05-16:
I can't find how to increase the laser power programmatically, when start camera. @RealSense-Customer-Engineering Any reference? It should be in json config?

RealSense-Customer-Engineering, 2018-05-16:
[Realsense Customer Engineering Team Comment]
@dmitryshendryk
you can adjust the laser power by using the set_option and get_option_range API.

    rs2::context ctx;
        // Declare RealSense pipeline, encapsulating the actual device and sensors
        rs2::pipeline pipe(ctx);

        // Start streaming with default recommended configuration
        pipe.start();
     rs2::device dev = pipe.get_active_profile().get_device();
     rs2::option_range range = dev.query_sensors()[0].get_option_range(rs2_option::RS2_OPTION_EXPOSURE);
     dev.query_sensors()[0].set_option(rs2_option::RS2_OPTION_EXPOSURE, range.min + range.step);

RealSense-Customer-Engineering, 2018-05-16:
[Realsense Customer Engineering Team Comment]
@dmitryshendryk
Please change the exposure enum value to **rs2_option::RS2_OPTION_LASER_POWER** to change the laser power, the code below is for changing the exposure.

dmitryshendryk, 2018-05-16:
I use rs2::sensor, not pipe, because we need callbacks for frame processing. For camera setup I use json config. Thanks for replies

closed 2018-05-16 by dmitryshendryk