realsenseai/librealsense #1700: Support Ubuntu 18.04

feliwir, 2018-05-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415 | 
| Firmware Version                   | 5.08.15.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-20-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.11.0                        | 

### Issue Description
Hello, the issue i am having that i am unable to update the kernel drivers to 4.15. Running the script gives me an error that kernel version 4.15 isn't supported. Since this is a LTS release it would be worth adding this kernel version in my opinion. It's also not possible for me to update the firmware because the debian repositories are down. Thanks!

ev-mp, 2018-05-15:
Hello @feliwir ,
We are working to add kernel 4.15 support for Ubuntu 18 and it is scheduled for the next version release.
We will also look into updating the debian repositories due to transition from xenial to bionic.

In the meanwhile you're welcomed to use the kernel patch for Ubuntu 18 from WIP branch
https://github.com/ev-mp/librealsense/tree/kernel_4.15

RealSense-Customer-Engineering, 2018-05-23:
[Realsense Customer Engineering Team Comment]
Hi @feliwir,

Is the issue resolved? You can also refer to below for Ubuntu 18 installation instruction.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

feliwir, 2018-05-23:
@RealSense-Customer-Engineering 
Hello, according to the installation instructions this still doesn’t seem possible with Kernel version 4.15, which is the default to Ubuntu 18.04

ev-mp, 2018-05-23:
Hello @feliwir ,
Ubuntu 18 with kernel 4.15 is supported. It was not listed in one of the preparation steps (updated now), so it could confuse. 

Please follow the guide to build kernel patches and update if you have issues.

feliwir, 2018-05-23:
Hey @ev-mp ,
when running the patch script i need to click through a lot of choices with enter myself. After selecting all the options i get the following:
`Replacingvideodev :
	Module videodev  is used by uvcvideo
	Unloading dependency uvcvideo
	Module is resident, unloading ...  succeeded. 
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Exec format error
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded
`

ev-mp, 2018-05-23:
Please post the kernel log (`dmesg`) and the script log for review

feliwir, 2018-05-24:
@ev-mp i guess we meant the output of the patch script. The issue is that i can't capture that with the `>` operator because i always need to press enter manually. However here is the dmesg log:
[dmesg.txt](https://github.com/IntelRealSense/librealsense/files/2034613/dmesg.txt)

ev-mp, 2018-05-24:
@feliwir , thanks for the log.
It seem as if the kernel you're running was built without "Meltdown"-required patches. Seem a little strange as it was just one of the latest features both introduced and retrofitted into all the latest LTS kernels. 

Are you using Ubuntu 18.04 or Ubuntu 18.10 distribution?
Additionally, please check and provide host build and toolchain info:
`gcc -v`
`cat /sys/devices/system/cpu/vulnerabilities/spectre_v2`
'zgrep CONFIG_RETPOLINE /boot/config* /proc/config.gz'

feliwir, 2018-05-24:
Hey @ev-mp ,

I am using Ubuntu 18.04. I downloaded the .iso from the official website after it was officially released (so no beta version whatsoever)

gcc -v
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 8-20180414-1ubuntu2' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --with-as=/usr/bin/x86_64-linux-gnu-as --with-ld=/usr/bin/x86_64-linux-gnu-ld --program-suffix=-8 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 8.0.1 20180414 (experimental) [trunk revision 259383] (Ubuntu 8-20180414-1ubuntu2) 
```
cat /sys/devices/system/cpu/vulnerabilities/spectre_v2
```
Mitigation: Full AMD retpoline, IBPB
```
grep CONFIG_RETPOLINE /boot/config* /proc/config.gz
```
/boot/config-4.15.0-20-generic:CONFIG_RETPOLINE=y
/boot/config-4.15.0-22-generic:CONFIG_RETPOLINE=y
gzip: /proc/config.gz: No such file or directory
```

ev-mp, 2018-05-24:
I see recognize some differences:
1.The GCC versions differ
>gcc version 8.0.1 20180414 (experimental) [trunk revision 259383] (Ubuntu 8-20180414-1ubuntu2) 

while I have `gcc version 7.3.0 (Ubuntu 7.3.0-16ubuntu3)`.

2 Additionally
>Mitigation: Full AMD retpoline, IBPB

vs (local version)
`Mitigation: Full generic retpoline`

We will look into this to understand what are the requirements/constrain  to use gcc8-based tool-chain.
In the meantime, can you reconfigure the toolchain rebuild the kernel patches using gcc 7  ? ([this link](https://askubuntu.com/questions/1028601/install-gcc-8-only-on-ubuntu-18-04?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa) might be helpful)
Also, can you send a link to iso that you're using ?

feliwir, 2018-05-24:
Hey @ev-mp,
i downloaded the 64-bit .iso from here: http://releases.ubuntu.com/bionic/ (desktop of course). I installed gcc 8 via `apt install`, but not sure if i can switch the gcc version since this computer is used for development in work.

lramati, 2018-05-24:
I can't recall where, but I remember finding instructions online for how to
install multiple versions of gcc side by side. Could that be a solution for
you?

feliwir, 2018-05-24:
Also tried with gcc 7.3 which made no difference in the result

jecyrd, 2018-05-26:
Per the instructions contained in https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md

It says to run the script: ./scripts/patch-realsense-ubuntu-lts.sh
Starting on line 34 of the script, we have:

> ubuntu_codename=`. /etc/os-release; echo ${UBUNTU_CODENAME/*, /}`
if [ -z "$UBUNTU_CODENAME" ];
then
	# Trusty Tahr shall use xenial code base
	ubuntu_codename="xenial"
	retpoline_retrofit=1
fi
>
>kernel_name="ubuntu-${ubuntu_codename}-$kernel_branch"
>
>#Distribution-specific packages
if [ ${ubuntu_codename} == "bionic" ];
then
	require_package libelf-dev
	require_package elfutils
fi

The line first line correctly evaluates to bionic. The second line however evaluates to true as long as $UBUNTU_CODENAME isn't emtpy, and then it resets $ubuntu_codename to xenial. Then down on line 54, it pulls the xenial kernel rather than the bionic kernel from git://kernel.ubuntu.com. 

For me, after commenting out that if block (lines 35 to 40),  the patch runs successfully. I have not yet finished the installation process to see if everything else works.

ev-mp, 2018-05-27:
Hello @jecyrd , it is not directly related to the original issue, but
>The second line however evaluates to true as long as $UBUNTU_CODENAME **isn't emtpy**...

According to multiple resources (e.g.  [1](http://tldp.org/LDP/abs/html/comparison-ops.html), [2](https://www.cyberciti.biz/faq/unix-linux-bash-script-check-if-variable-is-empty/) ) the operator `if [ -z "$var" ] ` evaluates to true when the variable is unset or empty.

Can you share the execution log for lines 34-40 when you run unmodified script ? (enable log by replacing [`set -e`](https://github.com/IntelRealSense/librealsense/blob/development/scripts/patch-realsense-ubuntu-lts.sh#L4) with `set -ex`)

jecyrd, 2018-05-27:
Sorry. Suppose I didn't read the thread close enough. Though I was having the same modprobe error as feliwir and was avoiding starting a duplicate. 

> According to multiple resources (e.g. 1, 2 ) the operator if [ -z "$var" ] evaluates to true when the variable is unset or empty

You are correct, I got it turned around. The behavior is as I described, just for the wrong reasons (see script output below). I think the issue is that the condition needs to `[ -z "$ubuntu_codename"]` rather than `[ -z "$UBUNTU_CODENAME"]`. Which also worked for me, as an alternative to commenting out the if block.

Output of unmodified script:
```
$ ./scripts/patch-realsense-ubuntu-lts.sh 
+ exec
+ exec
+ '[' 1 -ne 0 ']'
+ echo -e '\e[32m'

+ read -p 'Remove all RealSense cameras attached. Hit any key when ready'
Remove all RealSense cameras attached. Hit any key when ready
+ echo -e '\e[0m'

+ source ./scripts/patch-utils.sh
+ sudo apt-get install linux-headers-generic build-essential git
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).
git is already the newest version (1:2.17.0-1ubuntu1).
linux-headers-generic is already the newest version (4.15.0.22.23).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ require_package libusb-1.0-0-dev
+ package_name=libusb-1.0-0-dev
+ printf '\e[32mPackage required %s: \e[0m' libusb-1.0-0-dev
Package required libusb-1.0-0-dev: + exec
+ exec
+ set -e
+ '[' 1 -eq 0 ']'
+ echo -e '\e[32m - found\e[0m'
 - found
+ require_package libssl-dev
+ package_name=libssl-dev
+ printf '\e[32mPackage required %s: \e[0m' libssl-dev
Package required libssl-dev: + exec
+ exec
+ set -e
+ '[' 1 -eq 0 ']'
+ echo -e '\e[32m - found\e[0m'
 - found
+ retpoline_retrofit=0
++ uname -r
+ LINUX_BRANCH=4.15.0-22-generic
++ choose_kernel_branch 4.15.0-22-generic
++ IFS=.
++ read -a kernel_version
++ case "${kernel_version[1]}" in
++ echo master
+ kernel_branch=master
++ . /etc/os-release
+++ NAME=Ubuntu
+++ VERSION='18.04 LTS (Bionic Beaver)'
+++ ID=ubuntu
+++ ID_LIKE=debian
+++ PRETTY_NAME='Ubuntu 18.04 LTS'
+++ VERSION_ID=18.04
+++ HOME_URL=https://www.ubuntu.com/
+++ SUPPORT_URL=https://help.ubuntu.com/
+++ BUG_REPORT_URL=https://bugs.launchpad.net/ubuntu/
+++ PRIVACY_POLICY_URL=https://www.ubuntu.com/legal/terms-and-policies/privacy-policy
+++ VERSION_CODENAME=bionic
+++ UBUNTU_CODENAME=bionic
++ echo bionic
+ ubuntu_codename=bionic
+ '[' -z '' ']'
+ ubuntu_codename=xenial
+ retpoline_retrofit=1
+ kernel_name=ubuntu-xenial-master
+ '[' xenial == bionic ']'
+ '[' '!' -d ubuntu-xenial-master ']'
+ git clone -b master git://kernel.ubuntu.com/ubuntu/ubuntu-xenial.git --depth 1 ./ubuntu-xenial-master
Cloning into './ubuntu-xenial-master'...
```

ev-mp, 2018-05-27:
Hello @jecyrd ,
thanks for the log - indeed the condition is not evaluated properly resulting in the version being overwritten with `xenial`:
>+ '[' -z '' ']'
+ ubuntu_codename=xenial
+ retpoline_retrofit=1

 - I'm away from Ubuntu18, but the variable should probably be encapsulated with curly braces for robustness `[ -z "${UBUNTU_CODENAME}"]` - can you share the bash version for reproduction ? (`echo $BASH_VERSION`)

compiaffe, 2018-06-07:
I ran into the same modprobe error on PopOS 18.04.
```
Replacingvideodev :
	Module videodev  is used by uvcvideo
	Unloading dependency uvcvideo
	Module is resident, unloading ...  succeeded. 
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Exec format error
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded
```

For me `UBUNTU_CODENAME` on line 34 didn't evaluate to anything. I tried just evaluating it by itself  `echo ${UBUNTU_CODENAME}` on my 18.04 PopOS, as well as inside a ubuntu:18.04 docker container - neither worked.

My solution was simply setting an environment in the shell before executing the script:
```
export UBUNTU_CODENAME=bionic
./scripts/patch-realsense-ubuntu-lts.sh
```
Then it ran through smoothly.

armandok, 2018-06-19:
The solution from @compiaffe worked for me!

feliwir, 2018-06-25:
@RealSense-Customer-Engineering 
There still is no package for the Ubuntu 18.04 if i am not mistaken. Until this isn’t done i wouldn’t close this issue.

RealSense-Customer-Engineering, 2018-06-26:
[Realsense Customer Engineering Team Comment]
Hi @feliwir,

You're right. The DKMS package is not there yet. Will further check internally about its availability.

RealSense-Customer-Engineering, 2018-07-27:
[Realsense Customer Engineering Team Comment]
Please refer to below instructions and DKMS package ready for kernel 4.15.
https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md

RealSense-Customer-Engineering, 2018-08-02:
[Realsense Customer Engineering Team Comment]
Hi @feliwir,

If no other question for this issue, will close this first. If you still can re-open or create another ticket.

closed 2018-08-03 by RealSense-Customer-Engineering