realsenseai/librealsense #1703: Where I can find a librealsense with version v1.12.1  ^_^

xufeifeiWHU, 2018-05-15:
| Required Info                                                                                           |
|-----------------------------------------------------------------------------                 |
| Camera Model                        | D415                                                       | 
| Firmware Version                    | (Open RealSense Viewer --> Click info) | 
| Operating System & Version   |   ubuntu 16.04 kernel                              | 
| Kernel Version (Linux Only)    |  4.13                                                        | 
| Platform                                   | PC                                                           |
| SDK Version                            |  not installed now                           | 

when i read the guide in  ../wrapper/python
it says:

    Note:

    pyrealsense AKA pyrealsense/2.0 is a community supported Python wrapper for librealsense v1.12.1, This wrapper does not support newer versions and does not work with the RealSense SDK 2.0.

Where I can find a librealsense with version v1.12.1  ^_^

ev-mp, 2018-05-15:
Version 1.12.1 is available for download on the 'legacy' branch [here](https://github.com/IntelRealSense/librealsense/tree/legacy).
Note that this version is intended primarily for owners of R200, LR200 and ZR300 devices and does not support the current line of D400 peripheral devices.

lramati, 2018-05-21:
Keep in mind that the note only applies to the PyPi pyrealsense package.

The binaries mentioned under the Windows Installation section, and the entirety of the Building From Source section are about pyrealsense2, which does work with the Realsense SDK 2.0

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
hi @xufeifeiWHU,

do you need more support on this topic?

closed 2018-06-15 by xufeifeiWHU