realsenseai/librealsense #1705: Converting bag file to video

Loriskar, 2018-05-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410 | 
| Firmware Version                   | Newest | 
| Operating System & Version |  Win10 /Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.4| 
| Platform                                 | PC  |
| SDK Version                            |  2.11.0      | 

I am planning to convert a bag file recorded by a D410 module into video (or jpeg for each frame), I am trying to follow the guide in ROS website(http://wiki.ros.org/rosbag/Tutorials/Exporting%20image%20and%20video%20data), and it is not working, I don't know if the device is not supported or we need a specific launch file design for d410 to acquire video. Or maybe it is a good idea to use the save to disk sample to do it frame by frame.

dorodnic, 2018-05-30:
Hi @Loriskar 
We recently added a new [utility](https://github.com/IntelRealSense/librealsense/tree/development/tools/convert) to the development branch to convert bag files into images. Let us know if it can help you.

marcovs, 2018-06-01:
Hi @dorodnic,

Thank you for the util - I tested the code but I get the same issue that I flagged in issue #1579, multiple calls to pause and resume will make the program crash. Does this happen at your end too?

Loriskar, 2018-06-03:
@dorodnic Thanks a lot

RealSense-Customer-Engineering, 2018-06-19:
[Realsense Customer Engineering Team Comment]
Hi @Loriskar,

Since the util is available there, will close this. If any more concern/question, you can file another issue. Thanks.

closed 2018-06-19 by RealSense-Customer-Engineering