realsenseai/librealsense #1706: how to get the intel D435 intrinsic and extrinsic calibration information

super-scyu, 2018-05-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

SimonLit, 2018-05-16:
Intrinsic information can be gatherd in the following way:  
       ` auto const i = pipe.get_active_profile().get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>().get_intrinsics();` Where pipe is of type `rs2::pipeline`. Change `RS2_STREAM_DEPTH` if you want intrinsic data from a different stream.  

Gathering extrinsic data can be done in the same way but replacing `get_intrinsics()` with `get_extrinsics_to(const rs2::stream_profile &to)`.

I don't know if this is the right way of doing it but it is a way to get the intrinsic and extrinsic data.

RealSense-Customer-Engineering, 2018-05-21:
[Realsense Customer Engineering Team Comment]
@super-scyuare
are those examples able to solve your issue?

super-scyu, 2018-05-24:
thankyou ,but i how to ootput intrinsics and extrinsics

super-scyu, 2018-05-24:
how to output intrinsics and extrinsics

super-scyu, 2018-05-24:
and i use the sensor-control example but there is no Distortion

RealSense-Customer-Engineering, 2018-06-05:
[Realsense Customer Engineering Team Comment]
Hi @super-scyu,
could the calibrationAPI package solve your problem?

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
Hi @super-scyu,
are the two package provided through below link can help on your need?
https://downloadcenter.intel.com/download/27415/Intel-RealSense-D400-Series-Dynamic-Calibration-Software-Tools

closed 2018-07-08 by dorodnic

narutojxl, 2019-10-25:
when launch D435 `roslaunch realsense2_camera rs_camera.launch`, there are some info topics about intrinsic, and some extrinsics topics about extrinsic params.  Launch urdf file `roslaunch realsense2_description  view_d435_model.launch` to see better.

> /camera/color/camera_info 
> /camera/depth/camera_info,  /camera/infra1/camera_info, /camera/infra2/camera_info, the 3 topics are the same.
>/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2