realsenseai/librealsense #1707: *Getting Started with OpenCV*'s example doesn't work on D415

longjj, 2018-05-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.13.0-38-generic          | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description

On Ubuntu 16.04 LTS, I can run `realsense-view` successfully. When I try the code in [Getting Started with OpenCV](https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md), **it throw an errror:**

```cpp
tanyang@pixtalks-013:~/projects/realsense$ ./BGR
terminate called after throwing an instance of 'rs2::error'
  what():  Failed to resolve request. No device found that satisfies all requirements
Aborted (core dumped)
```

It seems that the PC can't find the device, however, when I try the code in [rs400_advanced_mode.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400/rs400_advanced_mode.md), its output shows that my device has been connected:

```cpp
tanyang@pixtalks-013:~/projects/realsense$ ./main
There are 1 connected RealSense devices.
```

ev-mp, 2018-05-16:
Hello @longjj ,
Please run `realsense-viewer` to verify that D415 is recognized as USB3 and not USB2 device.

longjj, 2018-05-16:
@ev-mp  It seems that I have used USB3.0 interface cause I had read the Troubleshooting Q&A at the beginning, while I didn't verify it on `realsense-viewer`. I will check it tomorrow since the device is in my workplace. Thank you.

In addition, [ready-to-hack](https://github.com/IntelRealSense/librealsense#ready-to-hack) code works well.

RealSense-Customer-Engineering, 2018-05-16:
[Realsense Customer Engineering Team Comment]
I also didn't see this issue on my test configuration: kernel 4.13.0-41 + librealsense 2.11 + D415 FW 5.9.11.
The BGR opencv example can run normally to display the color image.

longjj, 2018-05-16:
@RealSense-Customer-Engineering 
My Opencv's version is 2.4, does that matter?
I am so appreciate for your help.

longjj, 2018-05-17:
@RealSense-Customer-Engineering 
Today I reinstalled my computer system and reinstalled `opencv 3.0` and  `librealsense` and then the code works! Maybe the cause is the mess dependence's environment since the previous environment was configured by another stuff.

Thanks a lot!

closed 2018-05-17 by longjj