realsenseai/librealsense #1712: 3d coordinates of pixels

tolgasaglik, 2018-05-16:
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version | Linux Ubuntu 16  | 
| Kernel Version (Linux Only)    |   4.13.0                                       | 
| Platform                                 | PC / ROS Kinetic  |
| SDK Version                            |  { 2.11.0 }                          | 

### Issue Description
I use SR300 with Openpose software(which provides 2d info of skeletal nodes). I would like to get z coordinates of matching (x,y) to know the position of skeletal nodes. How can I get the pixel coordinates in x,y,z format on ROS kinetic?

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
@tolgasaglik
did you try to get the x,y,z data from the topic pointcloud2 after launch the rs_rgbd.launch?
https://github.com/intel-ros/realsense#rgbd-point-cloud

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
@tolgasaglik
Here is another instruction for you to map the 3D point to depth frame coordinate.

`points` frame behaves like the depth frame it was created from: It contains the array of stacked rows.
`++` will bring you to the next point to the right, unless you reach of of row, then you will jump to start of next row. Some points are going to be `(0,0,0)`. We keep those to allow easy navigation within the point-cloud grid (you can access point corresponding to pixel (i, j) using `v[j * width() +i]`)

RealSense-Customer-Engineering, 2018-06-14:
[Realsense Customer Engineering Team Comment]
@tolgasaglik
Do you still have any further question? I will close this ticket first, if you still have question, please reopen it or create new ticket.

closed 2018-06-14 by RealSense-Customer-Engineering