realsenseai/librealsense #1721: Merge point clouds, export to ply

kostas01, 2018-05-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.11 | 
| Operating System & Version |  Win 10  |                                 | 
| Platform                                 | PC |
| SDK Version                            |  2                       | 

I am trying to do 3D reconstruction of a face.

Ideally, I would like to "merge" point clouds (with textures) from multiple consequent frames (as I move the camera around the object), then export the resulting object as PLY.

Currently in SDK and viewer tool I see only an option to export current single frame. Also I can't find an option to merge the point clouds.

Does that mean I would rather have to use an 3rd party library to achieve this result? If so, could you recommend one? OpenCV?

Jiloc, 2018-05-17:
I can suggest you taking a look at https://github.com/PointCloudLibrary/pcl and https://github.com/IntelVCL/Open3D . I ended up using the second one

dorodnic, 2018-05-21:
Hi @Jiloc 
I was also looking into using Open3D for ICP-type stitching. Can you share any of your findings / code?

RealSense-Customer-Engineering, 2018-06-25:
[Realsense Customer Engineering Team Comment]
hi @kostas01,

would like to know if Jiloc's suggestion works for you?

RealSense-Customer-Engineering, 2018-07-02:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, and please feel free to open new ticket if any question.

closed 2018-07-02 by RealSense-Customer-Engineering

sleepy-maker, 2018-12-19:
@dorodnic Is there any progress of this?