realsenseai/librealsense #1724: RuntimeError: API version mismatch: librealsense.so was compiled with API version 2.11.0 but the application was compiled with 2.9.1!

longjj, 2018-05-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |05.08.15.00 | 
| Operating System & Version |    Linux (Ubuntu 16.04 LTS) | 
| Kernel Version (Linux Only)    | 4.13.0-41-generic                                 | 
| Platform                                 | PC  |
| SDK Version                            |        2        | 

### Issue Description

I use `pip install pyrealsense2` to get the python wrapper.
When I run the [`quick start` example](https://pypi.org/project/pyrealsense2/#description) , I get the error below.
```python
Traceback (most recent call last):
  File "quick.py", line 2, in <module>
    pipe = rs.pipeline()
RuntimeError: API version mismatch: librealsense.so was compiled with API version 2.11.0 but the application was compiled with 2.9.1! Make sure correct version of the library is installed (make install)
```

I follow this [doc](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) to install sdk 2.0 on my system.

So do I just need to compile the sdk from the source to update the librealsense.so ? Or it is the matter caused by python wrapper?

RealSense-Customer-Engineering, 2018-05-18:
[Realsense Customer Engineering Team Comment]
Hi @longjj,

Please check below instructions and examples there.
https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python
https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python/examples

dorodnic, 2018-05-21:
I believe that the wrapper version currently available via `pip` is somewhat older. You can either fetch the older version of the SDK, or build the wrapper from source using the documentation above. 
We are planning to update the package, but currently it is not part of our regular release cycle.

longjj, 2018-05-24:
@dorodnic @RealSense-Customer-Engineering 

Thx, it does work!

---

Some related issue and docs may be useful for the new commer.

#1220 
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

closed 2018-05-24 by longjj

Daniel2401, 2019-09-20:
Required Info |  
-- | --
Camera Model | D435
Firmware Version | 05.11.06.250
Operating System & Version | Ubuntu 16.04.6 LTS
Kernel Version (Linux Only) | 4.15.0-64-generic
Platform | PC
SDK Version | 2

Hi,

I have the same problem, but with different versions.

`API version mismatch: librealsense.so was compiled with API version 2.28.0 but the application was compiled with 2.25.1! Make sure correct version of the library is installed (make install)`

I followed https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python but still the above error still occurs.

My steps were:
1. `mkdir reposerver`
2. `cd reposerver`
3. `git clone https://github.com/IntelRealSense/librealsense.git`
4. `cd librealsense`
5. `mkdir build`
6. `cd build`
7. `cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python`
8. `make -j4`
9. `sudo make install`
10. `export PYTHONPATH=$PYTHONPATH:/usr/local/lib`