realsenseai/librealsense #1725: D435，can use it to get the date of Hand position and posture？

BerryOne, 2018-05-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
@BerryOne
Assuming you were asking about features provide hand position and posture?
The answer is no. Current SDK only provides color/depth image related API, such as retrieve image, alignment, point cloud and camera configuration. There is no MW provides digested information, such as posture, facial landmarks etc.
You can leverage the openCV library to detect hand and posture.

RealSense-Customer-Engineering, 2018-06-08:
[Realsense Customer Engineering Team Comment]
@BerryOne
Any feedback? Or can I close the ticket?

closed 2018-06-14 by RealSense-Customer-Engineering