realsenseai/librealsense #1727: Post-processing filters removing Depth metadata

Jason420Git, 2018-05-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (5.9.11) | 
| Operating System & Version |   {Win 10 / Linux (Ubuntu 16) |                                   | 
| SDK Version                            |  { legacy / 2.11.0 }                          | 

### Issue Description
We have found that using the temporal post-processing filter strips the frame of the depth metadata. This causes a problem with our system as we check the timestamps to verify we have a new one and not a repeated frame. Is this intended functionality? The metadata fields all become 0. We have observed this on both Linux and Windows.

ev-mp, 2018-05-18:
Hello @Jason420Git , the metadata should be preserved.
This is probably a bug and we will look into this.
Thanks for reporting this.

dorodnic, 2018-07-12:
Merged now to the development branch, will be released soon

dorodnic, 2018-07-22:
Should be resolved in 2.14.0

closed 2018-07-22 by dorodnic