realsenseai/librealsense #1728: Cannot build current master on MacOS

krstns, 2018-05-18:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   MacOS  | 
| Platform                                 | PC  |
| SDK Version                            |  Latest commit 8145416 | 

### Issue Description
I'm getting errors when trying to build current master, I am not sure how come travis-c got passed around that. My previous version, from a month ago, built just fine.

```
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6051:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category == RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6077:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category == RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6103:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category == RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6051:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category == RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6114:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category == RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6077:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category == RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6153:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category != RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6103:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category == RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6114:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category == RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
/Users/krystian/projects/github/librealsense/common/model-views.cpp:6153:25: error: use of undeclared identifier 'RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED'
        if (category != RS2_NOTIFICATION_CATEGORY_FIRMWARE_UPDATE_RECOMMENDED)
                        ^
5 warnings and 7 errors generated.
5 warnings and 7 errors generated.
make[2]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/model-views.cpp.o] Error 1
make[1]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/model-views.cpp.o] Error 1
make[1]: *** [tools/depth-quality/CMakeFiles/rs-depth-quality.dir/all] Error 2
make: *** [all] Error 2
```

krstns, 2018-05-18:
I sure did...
I thought removing the build and cleaning up would solve the problem.
How can I solve this?

lramati, 2018-05-18:
I think `make uninstall` should do it

>

dorodnic, 2018-05-21:
Hi @krstns 
Please also check #1671 as I think this is a similar issue. You can either `sudo make uninstall` or just remove the files.

RealSense-Customer-Engineering, 2018-05-22:
[Realsense Customer Engineering Team Comment]
hi @krstns,

just verified again, i can build the master branch here, please let us know if uninstall works for you, thanks.

krstns, 2018-05-22:
Yeah, it seems that was fixed. I was not aware i did install, maybe missed it in xcode somewhere.

Thank you

closed 2018-05-22 by krstns