realsenseai/librealsense #1729: R200 installation problem on UpBoard Robotic Development Kit

gnom1gnom, 2018-05-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200 | 
| Firmware Version                   | | 
| Operating System & Version |  Ubuntu 16.04.4 LTS  | 
| Kernel Version (Linux Only)    |  4.4.0-2-upboard                                         | 
| Platform                                 | UpBaord (Intel RealSense Robotic Development Kit) |
| SDK Version                            |                           | 

Hello Community!
I'm trying to launch my RealSense Robotic Development Kit, but I'm stuck with the installation of the camera.
I followed the guide from https://01.org/developerjourney/recipe/intel-realsense-robotic-development-kit
But any example from `/librealsense/build/examples` results in
```
RealSense error calling rs2_pipeline_start(pipe:0x291def0):
    No device connected
```

ls /dev/video* and lsmod | grep -q uvcvideo && echo "Kernel module found."

sudo dmesg | tail -n 50 returns
```
[  161.949674] usb 2-1: new SuperSpeed USB device number 3 using xhci_hcd
[  161.967877] usb 2-1: New USB device found, idVendor=8086, idProduct=0a80
[  161.967889] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  161.967896] usb 2-1: Product: Intel RealSense 3D Camera R200
[  161.967901] usb 2-1: Manufacturer: Intel Corp
[  161.967906] usb 2-1: SerialNumber: SN_2521004043
[  161.969782] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[  161.972849] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[  161.975466] uvcvideo: Unable to create debugfs 2-3 directory.
[  161.978491] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[  161.980654] uvcvideo: Unable to create debugfs 2-3 directory.
[  161.981222] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.4/input/input8
```

The `uname -r` returns `4.4.0-2-upboard`.

What might be the problem? Should I apply the kernel patching scripts?

dorodnic, 2018-05-21:
Hi @gnom1gnom 
Please note that you need to fetch the [legacy branch](https://github.com/IntelRealSense/librealsense/tree/legacy) of librealsense. If you are building librealsense from source, add `git checkout legacy` after `git clone` or `git fetch`

gnom1gnom, 2018-05-22:
Thank you @dorodnic!

closed 2018-05-22 by gnom1gnom