realsenseai/librealsense #1730: Detect dropped frames via wait_for_frames() timeout

mattfelsen, 2018-05-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  v2.10.4                          | 

Hello! I'm trying to detect dropped frames. I'm currently looking at the timestamp delta between two frames to see if it's longer than the framerate (plus a little wiggle room), which is working fine. For example, for 30fps the delta should be 33ms, so I set it to 40ms.

I'm also noticing that `pipeline.wait_for_frames(40)` (using the same timeout duration as above) times out as well as expected. My issue is how can I differentiate this instance from other errors that cause `wait_for_frames()` to fail such as a device disconnection?

Looking at the source for librealsense, here is the line where that error is thrown:
https://github.com/IntelRealSense/librealsense/blob/v2.10.4/src/pipeline.cpp#L598

I would like to not rely on string comparisons for "Frame didn't arrived." I see that there is an timeout value in the notification enum here, but I'm not sure how to receive this notification, or if it's even thrown in this case:
https://github.com/IntelRealSense/librealsense/blob/v2.10.4/include/librealsense2/h/rs_types.h#L18

Is it possible that `wait_for_frames()` could instead do something like return an empty frameset rather than throw an error, or attach some enum value to the error that does get thrown?

Thank you!

dorodnic, 2018-05-21:
Hello @mattfelsen 
That's a great point, we are open to discuss this change (and either integrate into future release or help you make a pull-request)

lramati, 2018-05-21:
This could also help solve the issue of how to gracefully handle file
playback EOF

mattfelsen, 2018-05-21:
I spent some time looking into this today and I noticed a few things. I changed `wait_for_frames()` to return an empty frameset instead of throwing an error and saw the following:

1. If `wait_for_frames(40)` times out, it does not necessarily mean a frame was dropped. When I examine the deltas between frames, occasionally I see instances where where an empty frameset is returned, but the deltas for the few frames before & after are all around 33ms, so it doesn't look like anything was dropped
2. If a frame actually was dropped, it seems to be reliably picked up in the delta calculation when the next non-empty frameset arrives, i.e. on the times that there are empty framesets where frames were dropped, in the next frameset or 2 I'll see a delta of ~60-100ms
3. Calculating frame deltas can't reliably calculate how many frames were dropped. For example, if I calculate a delta of 85ms between two frames on a 30fps stream, does that mean 2 frames were dropped, or 3?

So it seems that trying to detect dropped frames via a timeout on `wait_for_frames()` is redundant since the deltas should pick it up, and unreliable as a counter since the deltas are not necessarily close to the multiples of 33 (i.e. 66, 99, 132, etc.). In the log below they are, though I have also seen instances where the deltas have been 84 (approx 33ms x 2.5), 152 (approx 33 x 4.6), 116 (33 x 3.5).

On a related note, I tried subscribing to the notification callback as shown in the example I linked to above, and can confirm that I don't receive that notification. 

Here's a sample log of output I was receiving:
```
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 66, 1526941311178.276123, color delta: 79, 1526941311175.531982
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 33, 1526941311211.778320, color delta: 00, 1526941311175.531982
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 00, 1526941311211.778320, color delta: 34, 1526941311209.915771
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 33, 1526941311244.786865, color delta: 34, 1526941311243.779785
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 34, 1526941311278.415527, color delta: 33, 1526941311276.971680
INFO  [DepthSensorRealSense::processData@488] Frameset size: 0
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 34, 1526941311312.677979, color delta: 37, 1526941311313.274414
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 35, 1526941311347.777832, color delta: 36, 1526941311349.374512
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 31, 1526941311378.620605, color delta: 29, 1526941311378.521484
INFO  [DepthSensorRealSense::processData@488] Frameset size: 0
ERROR [SensorListener::onSensorEvent@148] Color frame dropped. Frame time: 1526941312.000000, Delta: 56
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 33, 1526941311411.486816, color delta: 56, 1526941311434.537109
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 34, 1526941311445.910889, color delta: 00, 1526941311434.537109
INFO  [DepthSensorRealSense::processData@488] Frameset size: 0
INFO  [DepthSensorRealSense::processData@488] Frameset size: 0
ERROR [SensorListener::onSensorEvent@148] Color frame dropped. Frame time: 1526941312.000000, Delta: 86
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 33, 1526941311478.427246, color delta: 86, 1526941311520.519043
ERROR [SensorListener::onSensorEvent@144] Depth frame dropped. Frame time: 1526941312.000000, Delta: 68
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 68, 1526941311546.081299, color delta: 00, 1526941311520.519043
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 32, 1526941311578.767090, color delta: 00, 1526941311520.519043
ERROR [SensorListener::onSensorEvent@148] Color frame dropped. Frame time: 1526941312.000000, Delta: 55
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 00, 1526941311578.767090, color delta: 55, 1526941311575.876709
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 34, 1526941311612.190918, color delta: 00, 1526941311575.876709
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 00, 1526941311612.190918, color delta: 38, 1526941311613.048340
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 33, 1526941311645.680908, color delta: 33, 1526941311646.628662
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 35, 1526941311680.337891, color delta: 34, 1526941311680.832520
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 31, 1526941311711.479736, color delta: 33, 1526941311713.220703
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 34, 1526941311745.754883, color delta: 33, 1526941311746.967529
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 32, 1526941311777.954102, color delta: 34, 1526941311780.547852
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 35, 1526941311812.061279, color delta: 33, 1526941311813.789307
[... cut ...]
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 00, 1526941312411.202881, color delta: 29, 1526941312418.037842
INFO  [DepthSensorRealSense::processData@488] Frameset size: 0
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 33, 1526941312444.785889, color delta: 00, 1526941312418.037842
ERROR [SensorListener::onSensorEvent@148] Color frame dropped. Frame time: 1526941312.000000, Delta: 68
INFO  [DepthSensorRealSense::processData@488] Frameset size: 2, depth delta: 33, 1526941312477.876465, color delta: 68, 1526941312486.382080
```

dorodnic, 2018-07-12:
Hi @mattfelsen 
We created #2049 based on your feedback. 
Regarding the second part of your question, can you please double check that you can see `HARDWARE_TIMESTAMP` in Viewer (or via call to `get_timestamp_domain()`)
You should be able to assume the timestamp would be roughly divisible by the FPS gap. 
Sorry for long delay in response, hope it still can help.

RealSense-Customer-Engineering, 2018-07-16:
[Realsense Customer Engineering Team Comment]
@Mattfelsen,

Wait_for_frames returns bool value instead of throwing exception has been merged into development branch, please have a test and give me your feedback, thank you.
https://github.com/IntelRealSense/librealsense/pull/2055

closed 2018-07-26 by dorodnic