realsenseai/librealsense #1731: Downgrading Firmware version on R200 Camera

sandeand, 2018-05-18:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200 | 
| Firmware Version                   | 1.0.72.06 (Works), 1.0.72.10 (Broken) | 
| Operating System & Version |   Win10 / Linux Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.13.0-41                               | 
| SDK Version                            |  Legacy                          | 

### Issue Description
Hello, I was wondering if it is possible to downgrade the firmware version on a camera. I have two r200 cameras, one is functional and the other does not work at all. I am attempting to figure out the source of the problem, and the only discernible difference between them is the firmware version, so I would like to downgrade the firmware on the broken camera and see if that is the problem.

I attempted to use the DCM to just change the firmware version on Windows, and I don't know if I am not understanding how it works or if the issue is with the camera or my installation of the DCM.

RealSense-Customer-Engineering, 2018-05-23:
[Realsense Customer Engineering Team Comment]
Hi @sandeand,

Can you describe more about the issue? Did you have the issue on Ubuntu and Windows both? Can that broken R200 still recognized as a USB3 device?

sandeand, 2018-05-23:
Absolutely! I have the issue on both Windows and Linux, and the broken device was recognized on Linux (can't confirm Windows, but it was likely recognized as well). I was able to find out the firmware version that way. The way the problem manifests is an error "timeout waiting for frames", and when I investigated the forums for the issue I hit a dead end. This occurs even when I am using the example code from the library, and is only fixed by using a different camera.

RealSense-Customer-Engineering, 2018-05-25:
[Realsense Customer Engineering Team Comment]
Hi @sandeand,

Can you try below commands on that broken R200 and show me the logs:
1. $ lsusb
2. $ usb-devices
2. $ cpp-enumerate-devices (this app is built from librealsense1)

sandeand, 2018-05-25:
Here you go:
lsusb:
Bus 001 Device 002: ID 8087:8001 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 8086:0a80 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 8087:0a2a Intel Corp. 
Bus 002 Device 005: ID 03f0:0d4a Hewlett-Packard 
Bus 002 Device 004: ID 03f0:094a Hewlett-Packard Optical Mouse [672662-001]
Bus 002 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 002 Device 006: ID 0930:6508 Toshiba Corp. TravelDrive 2C
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

usb-devices:
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 2
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.13
S:  Manufacturer=Linux 4.13.0-43-generic ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=0000:00:1d.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480 MxCh= 8
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=8087 ProdID=8001 Rev=00.03
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh=11
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.13
S:  Manufacturer=Linux 4.13.0-43-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:00:14.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  6 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0930 ProdID=6508 Rev=01.00
S:  Product=USB Flash Memory
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 3 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage

T:  Bus=02 Lev=01 Prnt=01 Port=02 Cnt=02 Dev#=  2 Spd=480 MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1a40 ProdID=0101 Rev=01.11
S:  Product=USB 2.0 Hub
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=02 Lev=02 Prnt=02 Port=02 Cnt=01 Dev#=  4 Spd=1.5 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=03f0 ProdID=094a Rev=01.00
S:  Manufacturer=PixArt
S:  Product=HP USB Optical Mouse
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid

T:  Bus=02 Lev=02 Prnt=02 Port=03 Cnt=02 Dev#=  5 Spd=1.5 MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=03f0 ProdID=0d4a Rev=02.07
S:  Product=HP USB Keyboard
C:  #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid

T:  Bus=02 Lev=01 Prnt=01 Port=06 Cnt=03 Dev#=  3 Spd=12  MxCh= 0
D:  Ver= 2.01 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
[... 242 lines cut ...]
  628	x 468	@ 30Hz	Y16	56.3 x 43.5 degrees, distortion = NONE
  492	x 372	@ 90Hz	Y8	57.1 x 44.7 degrees, distortion = NONE
  492	x 372	@ 90Hz	Y16	57.1 x 44.7 degrees, distortion = NONE
  492	x 372	@ 60Hz	Y8	57.1 x 44.7 degrees, distortion = NONE
  492	x 372	@ 60Hz	Y16	57.1 x 44.7 degrees, distortion = NONE
  492	x 372	@ 30Hz	Y8	57.1 x 44.7 degrees, distortion = NONE
  492	x 372	@ 30Hz	Y16	57.1 x 44.7 degrees, distortion = NONE
  480	x 360	@ 90Hz	Y8	55.9 x 43.4 degrees, distortion = NONE
  480	x 360	@ 90Hz	Y16	55.9 x 43.4 degrees, distortion = NONE
  480	x 360	@ 60Hz	Y8	55.9 x 43.4 degrees, distortion = NONE
  480	x 360	@ 60Hz	Y16	55.9 x 43.4 degrees, distortion = NONE
  480	x 360	@ 30Hz	Y8	55.9 x 43.4 degrees, distortion = NONE
  480	x 360	@ 30Hz	Y16	55.9 x 43.4 degrees, distortion = NONE
  332	x 252	@ 90Hz	Y8	57 x 44.8 degrees, distortion = NONE
  332	x 252	@ 90Hz	Y16	57 x 44.8 degrees, distortion = NONE
  332	x 252	@ 60Hz	Y8	57 x 44.8 degrees, distortion = NONE
  332	x 252	@ 60Hz	Y16	57 x 44.8 degrees, distortion = NONE
  332	x 252	@ 30Hz	Y8	57 x 44.8 degrees, distortion = NONE
  332	x 252	@ 30Hz	Y16	57 x 44.8 degrees, distortion = NONE
  320	x 240	@ 90Hz	Y8	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 90Hz	Y16	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 60Hz	Y8	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 60Hz	Y16	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 30Hz	Y8	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 30Hz	Y16	55.2 x 42.8 degrees, distortion = NONE

RealSense-Customer-Engineering, 2018-05-29:
[Realsense Customer Engineering Team Comment]
Hi @sandeand,

It looks the broken R200 basically looks fine per the info.
Any other librealsense1 apps did you try to run on Linux or Windows? like 'cpp-config-ui' and 'cpp-capture'? These apps don't rely on DCM if they run on Windows. both show the error "timeout waiting for frames"?

RealSense-Customer-Engineering, 2018-05-30:
[Realsense Customer Engineering Team Comment]
Hi @sandeand,

1. For FW update, I suggest to re-program the broken R200 to make sure the firmware upgrade is normal. Try below to forcedly re-program the firmware:
Download the DCM (it should be 2.1.27.2853)
Run: intel_rs_dcm_r200_xxx.exe --extract-to=c:\temp --extract-only
Go to the C:\temp folder
Run: FWUpdateR200.exe -force -confirmed -y ../Data/DSFirmware/DS4_Firmware.ibin -f ../data/DSFirmware/DS4_Firmware.ihex
2. If above is still not working, I would think the broken R200's USB connector has something wrong. Due to many times cable connection, the USB receptacle is loose.

sandeand, 2018-06-07:
This worked, thank you for the assistance!

closed 2018-06-07 by sandeand