realsenseai/librealsense #1732: Can't open RealSense Viewer and Depth Quality Tool

adarvit, 2018-05-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC|
| SDK Version                            |   2.11.1 | 

<bug>

hey, since my windows has been updated to version 1803 i cant open RealSense Viewer and Depth Tool.
I tried to open them as administrator, and even used cmd line, but it didn't response.

In addition, when i open calibration tool the text is blur 

[image]

dorodnic, 2018-05-21:
Hi @adarvit 
Do you get any error message when trying to run the latest version of the Viewer from command line? 
In general, this has something to do with OpenGL drivers on your machine. The boxed-font issue usually is related to not sufficient graphic memory. I suspect the Viewer and DQT detect this and refuse to run because of it.

adarvit, 2018-05-21:
I dont get any error when i am trying to run it using command line

[image]

dorodnic, 2018-05-21:
What type of graphic card are you using? Did you try to download and install the latest drivers?

adarvit, 2018-05-21:
intel 620 and amd radeon 530
and yes, i have the latest drivers

closed 2018-05-21 by adarvit