realsenseai/librealsense #1733: How to transmit data between D435 and Qt?

Destinyzc, 2018-05-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.13.0                                     | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<10>.<4> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I want to design a UI interface with Qt to change the exposure and gain using spinBox like the simple RealSenseViewer. But when I want to transmit the pictures of D435 to Qt interface and deliver the data of spinBox to D435, I ran into trouble. Can someone give me a advice?
Thanks a lot.

Jiloc, 2018-05-22:
https://github.com/Jiloc/tighty here is simple example qt wrapper

RealSense-Customer-Engineering, 2018-06-11:
[Realsense Customer Engineering Team Comment]
hi @Destinyzc,

do you need more support on this topic?

closed 2018-06-21 by RealSense-Customer-Engineering