realsenseai/librealsense #1734: Is there  any Gimbal Stabilizer for intel realsense 400 camera?

xielhjia, 2018-05-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSense-Customer-Engineering, 2018-06-27:
[Realsense Customer Engineering Team Comment]
hi @xielhjia,

sorry for late, as I far I know, we have not tested or have a recommendation,   I am assuming the stabilizer is to hold the camera and compensate for shaking, please let me know if this is not your question.

closed 2018-06-28 by xielhjia

kubrakaracan, 2019-05-06:
Hi, 

I would like to use a gimbal stabilizer for D435, do you have any suggestion or experience with any gimbal? 

Thanks in advance!

KSanthanam, 2019-12-10:
I want to attach RealSense D435 to a drone on a gimbal so I can move remotely. Any suggestions, please.