realsenseai/librealsense #1739: Can

XMYFYM, 2018-05-22:
| Camera Model                       | D400  | 
| Firmware Version                   | don‘t konw | 
| Operating System & Version |   Win10| 
| Kernel Version (Linux Only)    |        | 
| SDK Version                            |  2.0 | 

I tried to call D415 through the matlab image collection box,  the result failed, is my operation problem or matlab call has not yet developed？

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
hi @XMYFYM,

yes, Matlab still on the way.

XMYFYM, 2018-05-28:
Is the SR300 the same?

jbarbosa29, 2018-06-04:
Any release date for matlab wrapper?

dorodnic, 2018-09-03:
We now provide [matlab bindings](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/readme.md) for the [Windows SDK](https://github.com/IntelRealSense/librealsense/releases/download/v2.16.0/Intel.RealSense.SDK.exe).

RealSense-Customer-Engineering, 2018-09-11:
[Realsense Customer Engineering Team Comment]
Hi @XMYFYM,

Did you try the matlab wrapper? Still need any support for this topic?

XMYFYM, 2018-09-12:
I haven't tried it yet. Do you mean that D415 can connect to matlab now?

RealSense-Customer-Engineering, 2018-09-14:
[Realsense Customer Engineering Team Comment]
Hi @XMYFYM,

Yes, please check below link.
https://github.com/IntelRealSense/librealsense/tree/development/wrappers/matlab

RealSense-Customer-Engineering, 2018-09-26:
[Realsense Customer Engineering Team Comment]
Hi @XMYFYM,

any update?

closed 2018-10-08 by RealSense-Customer-Engineering