realsenseai/librealsense #1741: CLR Windows Form Problem

tarekmuallim, 2018-05-22:
| Camera Model                       |        D400  | 
| Operating System & Version |          Windows 10 x64  (Visual Studio 2017) | 
| Windows SDK version   |       10.0.16299.0      | 
|  Platform toolset      |      Visual Studio 2017 (v141)     |
|  Platform    |      Visual Studio 2017 (v141)     |
| SDK Version                            |      2.10.4                          | 

I am trying to build a GUI using CLR Windows Form to get data from D415.
But as I include librealsense2/rs.hpp I get this error:
Error	C1189	#error:  <mutex> is not supported when compiling with /clr or /clr:pure.

How can I fix this problem?

RealSense-Customer-Engineering, 2018-05-28:
[Realsense Customer Engineering Team Comment]
@tarekmuallim
I've tried to create Visual C++ Windows Desktop application and include the librealsense2/rs.hpp as attachment, I didn't see the effor you mentioned. Can you attach your project for me to check? thank you.

carypai, 2018-05-28:
Attachment for above comment.
[image]

tarekmuallim, 2018-05-29:
@carypai 
@RealSense-Customer-Engineering 
Thank you for the replay.
Actually, I am developing a C++/CLI application
[https://msdn.microsoft.com/en-us/library/hh875053.aspx](url)
[Project1.zip](https://github.com/IntelRealSense/librealsense/files/2046980/Project1.zip)

 so I am using a CLR windows form.
Attached you find a copy of the project.
Thanks again.

tarekmuallim, 2018-05-29:
[image]

tarekmuallim, 2018-05-29:
[Project1.zip](https://github.com/IntelRealSense/librealsense/files/2046989/Project1.zip)

dorodnic, 2018-05-30:
Hi @tarekmuallim 
I can confirm that what you are trying to do is correct and should work. `rs_types.hpp` seem to have couple of includes that has no place there. I'll create a pull-request and it will be published in 2.12 release. 
You have couple of options until then:
1. Use the .NET wrapper assembly. You can consume our .NET wrapper from both C# and C++ CLI (and any other CLR language) If your end-goal is WinForms / WPF type GUI, using C# would probably make more sense. 
2. Use the C API (`#include <librealsense2\rs.h>`
3. Manually remove `#include <mutex>`, `#include <condition_variable>` and `#include <atomic>` from `rs_types.hpp`.

tarekmuallim, 2018-05-30:
Hi @dorodnic 

Thank you for the reply.
Right now I have to use C API.

I didn't find  `#include <mutex>` in `rs_types.hpp` .
I find` #include <condition_variable>` and ` #include <atomic>` 
I also try to remove this two includes but still have error.

If the problem is in `rs_types.hpp` could you please send me a copy of the modified one?

Thanks again.

tarekmuallim, 2018-05-30:
ok I see.
Thank you for your help.

tarekmuallim, 2018-06-01:
Ok
Thank you.

closed 2018-06-01 by RealSense-Customer-Engineering