realsenseai/librealsense #1743: Is no "hid-sensor-accel-3d.ko" and "hid-sensor-gyro-3d.ko" on Raspberry Pi a problem for RealSense to work?

tken-Interrobot, 2018-05-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version | Ubuntu MATE 16.04.2 LTS | 
| Kernel Version (Linux Only)    | 4.4.38-v7+ | 
| Platform                                 | Raspberry Pi 3 Model B Rev 1.2  |
| SDK Version                            |  v2.11.1  | 

| Additional Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| USB hub (*1)                       | BSH4A01 (Buffalo) | 
| Swap size                            |  1 GB  |
| Swapiness                            |  100  |

(*1) [Raspberry Pi 3 not detecting D400 camera #904](https://github.com/IntelRealSense/librealsense/issues/904#issuecomment-373328695)

### Issue Description

I can finally run "realsense-viewer" successfully after following the [document for Raspberry Pi 3](https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md) and modifying the patch script "patch-realsense-ubuntu-lts.sh" for Raspberry Pi kernel.

[image]

But I have a couple of clarifications.

1. Which D435 camera firmware version started to support Raspberry Pi 3 ?

First, I had "05.08.15.00" with it as the document says it supports 5.08.14.0 and on, but got the following error when runnig "realsense-viewer". I updated just the FW to 05.09.11.00 and then it started to run.
Is the error due to the FW version? If it's the case, you might want to update the document?

```
 21/05 16:46:11,102 WARNING [1826161216] (sensor.cpp:330) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 21/05 16:46:32,396 WARNING [1780470336] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 21/05 16:46:34,416 WARNING [1788858944] (backend-v4l2.cpp:866) Empty frame has arrived.
 21/05 16:46:34,416 WARNING [1788858944] (backend-v4l2.cpp:866) Empty frame has arrived.
 21/05 16:46:34,416 WARNING [1788858944] (backend-v4l2.cpp:866) Empty frame has arrived.
 21/05 16:46:34,417 WARNING [1788858944] (backend-v4l2.cpp:866) Empty frame has arrived.
 21/05 16:46:34,477 WARNING [1780470336] (backend-v4l2.cpp:866) Empty frame has arrived.
 21/05 16:46:34,477 WARNING [1780470336] (backend-v4l2.cpp:866) Empty frame has arrived.
 21/05 16:46:34,478 WARNING [1780470336] (backend-v4l2.cpp:866) Empty frame has arrived.
 21/05 16:46:34,478 WARNING [1780470336] (backend-v4l2.cpp:866) Empty frame has arrived.
 21/05 16:46:38,931 WARNING [1809384000] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:46:38,932 ERROR [1809384000] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:46:39,422 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:46:39,483 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:46:44,427 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:46:44,487 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:46:44,979 WARNING [1809384000] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:46:44,979 ERROR [1809384000] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:46:49,433 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:46:49,492 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:46:51,006 WARNING [1809384000] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:46:51,006 ERROR [1809384000] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:46:54,437 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:46:54,497 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:46:57,032 WARNING [1809384000] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:46:57,032 ERROR [1809384000] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:46:59,443 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:46:59,503 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:03,067 WARNING [1809384000] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:47:03,068 ERROR [1809384000] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:47:04,448 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:04,507 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:09,335 WARNING [1809384000] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:47:09,336 ERROR [1809384000] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:47:09,454 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:09,513 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:14,457 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:14,517 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:15,368 WARNING [1809384000] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:47:15,369 ERROR [1809384000] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:47:19,466 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:19,523 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:21,468 WARNING [1809384000] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:47:21,469 ERROR [1809384000] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:47:24,467 WARNING [1788858944] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:24,527 WARNING [1780470336] (backend-v4l2.cpp:883) Frames didn't arrived within 5 seconds
 21/05 16:47:26,563 WARNING [1954848768] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0x21b5590 stopped
 21/05 16:47:26,566 WARNING [1954848768] (archive.cpp:326) All frames from stream 0x0x21b5590 are now released by the user
 21/05 16:47:27,516 WARNING [1809384000] (types.cpp:57) get_xu(id=7) failed! Last Error: Input/output error
 21/05 16:47:27,516 ERROR [1809384000] (error-handling.cpp:66) Error during polling error handler: get_xu(id=7) failed! Last Error: Input/output error
```

2. No "hid-sensor-accel-3d.ko" and "hid-sensor-gyro-3d.ko"

The kernel of Ubuntu MATE for Raspberry Pi seems to have no modules "hid-sensor-accel-3d.ko" and "hid-sensor-gyro-3d.ko" as it is, and "patch-realsense-ubuntu-lts.sh" did not build them and install failed.
I have modified the script, temporarily removed them from being built or installed and the realsense-viewer seems to be working.

What functionalities of D435 or which API of librealsense would be limited due to that?
Do you recommend for me to enable config for those modules and build the kernel to include them?

ev-mp, 2018-05-22:
Hello @tken-Interrobot ,
The specified modules are not required for D435 model and can be commented out from the patch.
The script is intended for desktop installation using Ubuntu LTS kernel that has the specified modules in stock version.
As stated, Raspberry Pi is not an officially supported platform, but we will take a note to review and update the documentation.

It is not clear from the log what prompted the errors and what was the requested streaming profile , but the latest firmware versions 5.9+  include USB2 improvements, including supports of RGB sensor (D415/D435) and reliability enhancements.

tken-Interrobot, 2018-05-23:
Hello @ev-mp ,
Thank you for your prompt response. I understand it in every respect of your answer.
I close this.

closed 2018-05-23 by tken-Interrobot