realsenseai/librealsense #1745: how to use HW SYNC with multi cameras?

Yusnows, 2018-05-22:
I want to use multi cameras, and I try use the HW sync with them.

I follow the "multi camera white paper" to do somethings. but it seems not work at all. the cameras just behave themselves. 

I use the function ` rs2_send_and_receive_raw_data(pdev, maser_cmd, 254, NULL); ` to set a camera to be master. and ` rs2_send_and_receive_raw_data(pdev, slave_cmd, 25, NULL); ` to set the other to be slave. and `master_cmd ` is set to be an array `uint8_t maser_cmd[25] = { 0x0,0x14,0x0,0xAB,0xCD,0x64,0x0,0x0,0x0,0x01,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0 }; ` , ` slave_cmd ` is ` uint8_t slave_cmd[25] = { 0x0,0x14,0x0,0xAB,0xCD,0x64,0x0,0x0,0x0,0x02,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0 }; `.

I also use an  oscilloscope to see if the master camera's sync pin output any signal. the fact is it didn't.

the multi camera white paper that I mention above is [this](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Multiple_Camera_WhitePaper.pdf)

Anyone have done this? please help.

Thank you!

agrunnet, 2018-05-22:
You do need special Firmware for this. We are working on an official release.

Yusnows, 2018-05-23:
Where can I get the special Firmware? is there any guidance?

RealSense-Customer-Engineering, 2018-05-25:
[Realsense Customer Engineering Team Comment]
hi @Yusnows,

it's not released yet, we will update when released.

mutp, 2018-05-25:
Looking forward to this firmware release as well. Is there a time estimate you can give us?

zuru, 2018-06-18:
Should be sometime in Q3 i think.

hflsdupont, 2018-07-17:
On API Update page, it says "Applies to rolling-shutter SKUs with FW 5.9.14+." Is it saying that D435 still cannot use HW sync for multicameras?

mutp, 2018-07-18:
Yes, this is only for D415. This is what it says [here](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense-D400-Series-Spec-Update.pdf).

[image]

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @Yusnows,
are you using D435? or do you need more support on this?

closed 2018-08-21 by RealSense-Customer-Engineering